cornell-zhang/heterocl #114: Segmentation fault (core dumped): hcl.compute with placeholders whose dim more than 2

tonyjie, 2019-07-01:
I'm trying to write a conv_layer using HeteroCL, but a memory error can't be solved. 
```
input = hcl.placeholder((4,32,10,10), "input")
filter = hcl.placeholder((32,32,3,3), "filter")
bias = hcl.placeholder((32,), "bias")
print(bias.shape)

def conv_layer(input, filter, bias):
    def compare_with_zero(a):
        with hcl.if_(a > 0):
            hcl.return_(a)
        with hcl.else_():
            hcl.return_(0)
    output = hcl.compute(input.shape, lambda n,c,h,w: compare_with_zero(input[n,c,h,w]), "output")
    return output
s = hcl.create_schedule([input, filter, bias], conv_layer)
print(hcl.lower(s))
f = hcl.build(s)
print('done')
```
Above program went error when doing with f = hcl.build(s)
However, when the input placeholder's dimension is 2, this program works. And when the dimension becomes 3 or 4, the error of Segmentation Fault (core dumped) came.

seanlatias, 2019-07-02:
HeteroCL does not support ``return_`` statement inside a normal Python function call. In this case, you have several options.

1. Use ``hcl.select``
```python
def compare_with_zero(a):
    return hcl.select(a > 0, a ,0)
```

2. Use an intermediate variable to hold the result
```python
def compare_with_zero(a):
    b = hcl.local(0)
    with hcl.if_(a > 0):
        b[0] = a
    with hcl.else_():
        b[0] = 0
    return b[0]
````

Please close this issue if this solves the problem.

tonyjie, 2019-07-02:
> HeteroCL does not support `return_` statement inside a normal Python function call. In this case, you have several options.
> 
> 1. Use `hcl.select`
> 
> ```python
> def compare_with_zero(a):
>     return hcl.select(a > 0, a ,0)
> ```
> 
> 1. Use an intermediate variable to hold the result
> 
> ```python
> def compare_with_zero(a):
>     b = hcl.local(0)
>     with hcl.if_(a > 0):
>         b[0] = a
>     with hcl.else_():
>         b[0] = 0
>     return b[0]
> ```
> 
> Please close this issue if this solves the problem.

Thanks! Both of the methods can work. In this case, I think the second option is a general solution for now, using a local intermediate variable to hold the result. 
By the way, another question is that when using return instead of hcl.return_(), the normal Python function didn't work. To be specific:
```
def conv_layer(input, filter, bias):
    def compare_with_zero(a):
        with hcl.if_(a > 0):
            return a
        with hcl.else_():
            return 0
```
when printing `hcl.lower(s)' , this function only became an `output = input`. That's the reason why I'm using `hcl.return_()` . I think the reason is that when we use some dsl function of hcl, such as `hcl.if_()` and `hcl.else_()`, HeteroCL doesn't support normal return. I'm not sure is that correct? 

Anyway, the respond helps. Thanks

seanlatias, 2019-07-02:
Yes, you are right. It would be a good idea to support ``hcl.return_`` within an if/else statement in the future.

closed 2019-07-02 by tonyjie

seanlatias, 2019-07-08:
Fixed by #116.