cornell-zhang/heterocl #115: Data Type Error if I don't use hcl.init(dtype)

tonyjie, 2019-07-02:
The error message is:
```
heterocl.tvm._ffi.base.TVMError: [13:55:02] src/codegen/llvm/llvm_module.cc:59: Check failed: ret == 0 (-1 vs. 0) Assert fail: ((((tvm_struct_get(arg0, 0, 5) == (uint8)2) && (tvm_struct_get(arg0, 0, 6) == (uint8)32)) && (tvm_struct_get(arg0, 0, 8) == (uint8)0)) && (tvm_struct_get(arg0, 0, 7) == (uint8)1)), arg0.dtype is expected to be float32
```
You can reproduce it by the code below:
```
hcl.init()
input = hcl.placeholder((1280, 768, 3), "input", dtype = hcl.Float()) #input shape = output shape
def bug(input_tensor):
    return hcl.compute(input_tensor.shape, lambda x,y,c: input_tensor[x,y,c], dtype=hcl.Float())

hcl.build(hcl.create_schedule([input], bug))(hcl.asarray(np.zeros(input.shape, dtype=float)), hcl.asarray(np.zeros(input.shape, dtype=float)))
```
I don't know what is arg0, because I think all the data type is set to hcl.Float(), and it can be shown by IR presented. 
This error can be solved by replacing `hcl.init()` to `hcl.init(hcl.Float())`. But I don't know where the wrong data type is. 

Thanks!

seanlatias, 2019-07-03:
The problem is that you need to specify the data type of the HeteroCL array instead of the Numpy array. For example,

``hcl.asarray(np.zeros(input.shape), dtype=hcl.Float())``

The data type set by ``hcl.init()`` has the highest priority. Maybe I can change this setting later but I need to think more.

tonyjie, 2019-07-03:
> The problem is that you need to specify the data type of the HeteroCL array instead of the Numpy array. For example,
> 
> ```python
> hcl.asarray(np.zeros(input.shape), dtype=hcl.Float())
> ```
> 
> The data type set by `hcl.init()` has the highest priority. Maybe I can change this setting later but I need to think more.

It works. Thank you for helping me fix my bugs lol. I didn't find it because I think the IR generated is the same... When I specify the data type of numpy array, the IR said `allocate sRGBtoLinear[float32 * 1280 * 768 * 3]`

Thanks a lot!

closed 2019-07-03 by seanlatias