cornell-zhang/heterocl #117: Tools version

faku99, 2019-07-10:
Hi

I would like to know if there are any requirements about the version of Vivado and Intel tools to use with HeteroCL?

Thanks for the answer and keep up the good work!

seanlatias, 2019-07-10:
We do not have a requirement for the tool version but following are the versions we have tested.

For Xilinx's flow we use Vivado & Vivado HLS 2017.2. We have also tested on Amazon F2 instance.
For Intel's flow we use i++ 17.1 & Quartus 17.0.

chhzh123, 2023-02-23:
As we have released a [new version](https://github.com/cornell-zhang/heterocl/releases/tag/v0.5) of HeteroCL that entirely moves the compilation stack to MLIR, issues related to the original HalideIR flow will be closed. Please check out the latest HeteroCL and see if the issue still exists.

For HeteroCL v0.5, we mainly support Vivado HLS 2019.2, and have initial support for Vitis HLS. Please feel free to have a try.

closed 2023-02-23 by chhzh123