cornell-zhang/heterocl #118: Create scheme from MXNet model

faku99, 2019-07-15:
Is there a way to create a scheme from an imported MXNet model without having to define it?

Something like:

```python
sym, arg, aux = onnx_mxnet.import_model(ONNX_MODEL_PATH)

hcl.create_scheme(arg, sym)
```

or do I have to define the network architecture from scratch as in the [LeNet inference tutorial](http://heterocl.csl.cornell.edu/doc/samples/lenet/lenet_main.html)?

seanlatias, 2019-07-15:
For now you need to build the network manually as in the LeNet tutorial. However, we are working on enabling automatic import from other ML frameworks. This will be included in our next release.

faku99, 2019-07-16:
Is there a release date planned for the next release?

seanlatias, 2019-07-22:
We plan to release it in fall.

seanlatias, 2020-08-31:
#142

closed 2020-08-31 by seanlatias