cornell-zhang/heterocl #119: Add bias_add operator

faku99, 2019-07-16:
I can't find a way to implement `bias_add` operator.

Let's say I have a `Tensor` with a shape of `(1, 8, 28, 28)` and another `Tensor` (which contains biases) with a shape `(8, 1, 1)`. How could I add the biases to the first `Tensor`?

TVM implements such an operator with C++ but I can't find a way to implement it with the `hcl.compute` API. Any thoughts?

seanlatias, 2019-07-16:
The easiest way to do that is using our [imperative DSL](http://heterocl.csl.cornell.edu/doc/tutorials/tutorial_02_imperative.html).

chhzh123, 2023-02-23:
As we have released a [new version](https://github.com/cornell-zhang/heterocl/releases/tag/v0.5) of HeteroCL that entirely moves the compilation stack to MLIR, issues related to the original HalideIR flow will be closed. Please check out the latest HeteroCL and see if the issue still exists.

closed 2023-02-23 by chhzh123