cornell-zhang/heterocl #122: Error when called 'hcl.power(2.7, _input[y, x])'

tonyjie, 2019-08-15:
Error when I try to use `hcl.power(2.71828, _input[y, x])`
Error information is `'float' object has no attribute 'dtype'`, so it seems that I can't use a simple number as the input argument. 
I can only bypass this problems by writing like that:
```
__expo = hcl.local(init = 2.71828, dtype = hcl.Float())
......
hcl.power(__expo[0], _input[y, x])
```
But the HeteroCL IR generated will cause some conflicts with what I'm doing on my project now...
So I'm wondering is it a way to use number as input argument of hcl.power function?

Thanks!

seanlatias, 2019-08-15:
Can you show me the IR and briefly explain what's the problem? Thank you.

tonyjie, 2019-08-15:
```
import heterocl as hcl
hcl.init()
__expo = hcl.local(init = 2.71828, dtype = hcl.Float())

def _all(_input, ):
    def test(y, x, _input):
        # return hcl.power(2.71828, _input[y, x])
        return hcl.power(__expo[0], _input[y, x])
    _output = hcl.compute((4, 4), lambda y, x : test(y, x, _input, ), name = "test_local")
    return _output

_input = hcl.placeholder((4, 4, ), name = "_input")
s = hcl.create_schedule([_input, ], _all)
f = hcl.build(s)
print(hcl.lower(s))
print(hcl.build(s, target = "soda"))
```
The HeteroCL IR is like:
```
// attr [local0] storage_scope = "global"
allocate local0[float32 * 1]
produce local0 {
  // attr [0] extern_scope = 0
  local0[0] = 2.718280f
}
produce test_local {
  // attr [0] extern_scope = 0
  for (y, 0, 4) {
    for (x, 0, 4) {
      test_local[(x + (y*4))] = int32(pow(local0[0], _input[(x + (y*4))]))
    }
  }
}
```

Above is a test program. Now it can work but can't generate soda backend code, because of the index of "local[0]" in HeteroCL IR is not affine. 

The ideal way is like the commented line `return hcl.power(2.71828, _input[y, x])`, and in HeteroCL IR replacing 'local[0]' with '2.71828'. After all, the original idea is not to set this number as a buffer 'local[0]'.

seanlatias, 2019-08-15:
I can provide a temporary solution. Instead of using "hcl.power", use the following
``hcl.call_pure_intrin("float", "power", 2.718280, _input[y, x])``
I will enable the support when I'm available. Let me know if this does not work.

tonyjie, 2019-08-15:
OK, now it works, but with little change of the code given:
`hcl.call_pure_intrin("float", "pow", 2.718280, _input[y, x])`

Thanks! Also please let me know when you enable the support. But this temporary solution seems enough for now.

seanlatias, 2019-09-17:
Should be fixed by #131.

closed 2019-09-17 by seanlatias