cornell-zhang/heterocl #124: Is there a struct-like data type in HeteroCL?

tonyjie, 2019-08-21:
I'm working on generating HeteroCL code from Halide IR. And I found it's hard to compute two value in one For loop in HeteroCL. To be more specific, following is a piece of Halide code:
```
RDom search(0, 64);
offset(x, y) = {cast<int8_t>(0), cast<uint16_t>(65535)};
offset(x, y) = {select(SAD(x, y, search.x) < offset(x, y)[1],
                      cast<int8_t>(search.x),
                      offset(x, y)[0]),
                     min(SAD(x, y, search.x), offset(x, y)[1])};
```
This code is to do a argmin computation, and need some iteration in it. So it need to compute two variables `offset(x, y)[0]` and `offset(x, y)[1]` in one For loop which can't be separated. 

I know I can actually increase a dimension (size = 2) to store the value, and do some similar iteration computation (I haven't tried if it works), but doing so will change the shape of output. Also, it's not the original meaning of the code. 

So I'm just wondering and asking is there a feature in HeteroCL to create a structure or a tuple-like data type, because I didn't find it successfully. 

Thnaks!

seanlatias, 2019-08-21:
Unfortunately no. The only solution right now is what you proposed. Namely, create a tensor with hyper dimension. I think since we use Halide as our back end, we should be able to solve this problem.

tonyjie, 2019-08-22:
Okay, get it. Thanks.

closed 2019-08-22 by tonyjie