cornell-zhang/heterocl #127: Integer Division Truncation is different to Halide

tonyjie, 2019-08-30:
I just found that HeteroCL's Integer division truncation is toward zero (-3 / 4 = 0), while the Halide behavior is truncation toward the lower integer (-3 / 4 = -1). This causes lots of problem when comparing the results between Halide programs and HeteroCL programs, because if there's negative integer division occured in the process, the results can't be the same...
So I'm wondering if there's a way to change this behavior to "Truncation toward Lower Integer"? 
Thanks!

seanlatias, 2019-08-30:
I think I'll need to add a mode to allow that because in C++ -3/4 is 0. I'll mark this issue as an enhancement.

seanlatias, 2019-08-30:
Actually, this is a good point because for Python -3/4 = -1.

tonyjie, 2019-08-30:
Yes. So I think for now I can't change this behavior, right? Looking forward to the upgrade.

chhzh123, 2023-02-23:
As we have released a [new version](https://github.com/cornell-zhang/heterocl/releases/tag/v0.5) of HeteroCL that entirely moves the compilation stack to MLIR, issues related to the original HalideIR flow will be closed. Please check out the latest HeteroCL and see if the issue still exists.

closed 2023-02-23 by chhzh123