cornell-zhang/heterocl #129: Typing Rules in HeteroCL Arithmetic

Blaok, 2019-09-04:
Do we have documentation about the typing rules in HeteroCL arithmetic? For example, it seems that the result of multiplication of two `int16` numbers is `int32`, which is inconsistent with [C++ behavior](https://en.cppreference.com/w/cpp/language/operator_arithmetic). It makes sense to extend the bitwidth and avoid overflow, but sometimes it can result in unnecessary hardware resource usage and thus may not always be the expected behavior. It might be better to allow users to turn this feature on and off, or even just document it properly and provide a way to bypass bitwith extenstion.

cc @tonyjie

seanlatias, 2019-09-05:
I believe what I've done is I will first cast up to maintain the precision then I cast down to the target precision. The reason that I did this is I want to make the CPU simulation results the same as Vivado HLS results. I agree with you that this may result in unnecessary hardware usage. So I think it makes sense to add an option to avoid this kind of automatic casting. Namely, we let the backend compilers to decide what to do.

chhzh123, 2023-02-23:
As we have released a [new version](https://github.com/cornell-zhang/heterocl/releases/tag/v0.5) of HeteroCL that entirely moves the compilation stack to MLIR, issues related to the original HalideIR flow will be closed. Please check out the latest HeteroCL and see if the issue still exists.

In the latest HeteroCL, we strictly follow how Vivado HLS implements their typing rules and have a extensible module for extending new typing rules for different operations.
https://github.com/cornell-zhang/heterocl/blob/main/heterocl/ast/type_rules.py

closed 2023-02-23 by chhzh123