cornell-zhang/heterocl #134: hcl.select(10>1, 0, 1) Error: First argument to Select is not a bool: int32

tonyjie, 2019-11-26:
I'm using hcl.select(cond, true, false) function, and just encounter a problem. 
When comparing two definite number, e.g. 10 > 1, the result is suppose to be a bool, but hcl.select() function doesn't think so......
It can be reproduced as follows:
```
import heterocl as hcl 
hcl.init()
def top(input, ):
    blur_x = hcl.compute((640, (480 + 2)), lambda x, y: 0, name = "blur_x", dtype = hcl.UInt(bits = 16))
    with hcl.Stage("blur_x"):
        with hcl.for_(0, 640, name = "x") as x:
            with hcl.for_(0, (480 + 2), name = "y") as y:
                blur_x[x, y] = ((input[(x + 2), y] + input[x, y] + input[(x + 1), y])/3) + hcl.select(10 > 1, 10, 1)
    return blur_x
input = hcl.placeholder((642, 482, ), name = "input", dtype = hcl.UInt(bits = 16))
s = hcl.create_schedule([input, ], top)
print(hcl.lower(s))
# f = hcl.build(s)
```
The error message is:
```
heterocl.tvm._ffi.base.TVMError: [01:36:00] src/ir/IR.cpp:140: Check failed: condition.type().is_bool() First argument to Select is not a bool: int32
```
If I change the `cond` in `hcl.select` from a definite number `10` to a variable `y`. Namely, `hcl.select(y > 1, 10, 1)`, it works. But I think based on Python syntax, the result of a comparison operation should be `bool` type, right?

seanlatias, 2019-11-26:
This is similar to #122, where the first argument should be a HeteroCL/TVM expression. I can do a quick fix.

seanlatias, 2019-11-26:
Should be fixed by #135. Let me know otherwise.

tonyjie, 2019-11-26:
The function `hcl.select(cond, true, false)' itself works well now. But I still have problems when I try to use this function as a part of index. It can be reproduced as follows. 
```
import heterocl as hcl 
hcl.init()
extent = 2
def top(input, ):
    blur_x = hcl.compute((640, (480 + 2)), lambda x, y: 0, name = "blur_x", dtype = hcl.UInt(bits = 16))
    blur_x[hcl.select(2 > 1, 0, 1), 0] = 2  
    return blur_x
input = hcl.placeholder((648, 482, ), name = "input", dtype = hcl.UInt(bits = 16))
s = hcl.create_schedule([input, ], top)
print(hcl.lower(s))
f = hcl.build(s)
```
The error message is:
```
    (...........)
    return _make.Select(condition, true_value, _make.Cast(true_value.dtype, false_value))
AttributeError: 'int' object has no attribute 'dtype'
```
Could you please solve this as well? Thanks!

seanlatias, 2019-11-27:
Please check #136. Also, you need to wrap your tensor assignment with a HeteroCL stage.

tonyjie, 2019-11-27:
Thanks. I think the problem is fixed now.

closed 2019-11-27 by tonyjie