cornell-zhang/heterocl #144: When would PolySA be released?

mikeseven, 2020-01-01:
PolySA is in roadmap for 0.3 since May.
When would HeteroCL integrate PolySA?

Thanks

whbldhwj, 2020-01-01:
Hi,

We are working on releasing a version at the end of February this year. Please check back for updates in March.

Best,

mikeseven, 2020-03-31:
Any news about PolySA?

whbldhwj, 2020-03-31:
Hi Mike,

Sorry for the delay. I've added many new features and am preparing for submission in late April. After the submission, the code will be open-sourced. I can share you with the under-development private repo on Github if you want.

Best regards and stay well!

mikeseven, 2020-04-19:
I'd love that.
Stay safe and well too!

whbldhwj, 2020-05-18:
Hi Mike,

Just let you know I just released an alpha version of PolySA (we renamed it to AutoSA).
https://github.com/UCLA-VAST/AutoSA
The tool is quite fragile now. You may take a look and let me know if you have any questions.

Best,

closed 2021-11-24 by Hecmay