cornell-zhang/heterocl #150: terminate called after throwing an instance of 'std::out_of_range'

Hecmay, 2020-01-21:
The error occurs when the user tries to create an `Allocate` statement without specifying the storage scope (some other issues can lead to the same error message). 

The complete error message from HetroCL is too ambiguous to pinpoint the issue:
```shell
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
Aborted
```
The error is caused by looking up a not existing entry in the map
```c++
const Variable* buffer = op->buffer_var.as<Variable>();
std::string scope = alloc_storage_scope_.at(buffer);
```
Adding a check statement before the lookup can provide more detailed error message
```c++
CHECK(alloc_storage_scope_.find(buffer) != alloc_storage_scope_.end());
```

seanlatias, 2020-01-21:
This issue is too ambiguous. Which file are you referring to? Which function raises the bug? Sometimes we write the logic in that way because we are certain that the ``.at`` will not cause any ``out_of_bound`` issue.

Hecmay, 2020-01-21:
The error happens in CodeGenC `tvm/src/codegen/codegen_c.cc`'s  IR Printer for Allocate. 

You are right, but when I created a reducer in the HeteroCL module, I did not see the AttrStmt attached to the Allocate statement of the reducer. And this leads to the SegFault, which took me quite a long time to locate where the error was located. Quite painful debugging experience to me...

zhangzhiru, 2020-01-21:
Let's add a regression test case for each bug. I suggest we create subfolders under 'tests'

seanlatias, 2020-01-22:
I agree, but this would require some work though. The reason is some code is from TVM's codebase. For example, the C++ code here was written by TVM. And unfortunately, we are out of sync with TVM. So even if TVM already fixed some bugs, we would not be able to reuse them.

seanlatias, 2020-08-31:
@Hecmay please see if the bug is still there.

Hecmay, 2020-08-31:
@seanlatias Already fixed.