cornell-zhang/heterocl #154: SegFault from `reuse_at` in HeteroCL module

Hecmay, 2020-02-04:
A simple minimal test case to reproduce the error.
```python
def kernel(A, B):
    @hcl.def_([(10,10), (10,10)])
    def test(A, B):
        hcl.update(B, lambda y, x: A[y,x] + A[y,x+1]+ A[y,x+2])
    test(A, B)

A = hcl.placeholder((10, 10))
B = hcl.placeholder((10, 10))
s = hcl.create_schedule([A, B], kernel)
s.reuse_at(kernel.test.A, s[kernel.test], kernel.test.axis[0])
```

The `reuse_at` primitive fails to locate the target buffer to be reused.

chhzh123, 2020-05-16:
This issue remains. I got a Segmentation Fault when running the example provided in [Memory Customization](http://heterocl.csl.cornell.edu/doc/tutorials/tutorial_06_memory.html).
```python
A = hcl.placeholder((6, 6), "A")
F = hcl.placeholder((3, 3), "F")

def kernel(A, F):
    r = hcl.reduce_axis(0, 3)
    c = hcl.reduce_axis(0, 3)
    return hcl.compute((4, 4),
            lambda y, x: hcl.sum(A[y+r, x+c]*F[r, c], axis=[r, c]), "B")

s_xy = hcl.create_schedule([A, F], kernel)
LB = s_xy.reuse_at(A, s_xy[kernel.B], kernel.B.axis[0], "LB")
WB = s_xy.reuse_at(LB, s_xy[kernel.B], kernel.B.axis[1], "WB")
print(hcl.lower(s_xy))
```

Hecmay, 2020-07-03:
Still have this issue when re-running optical flow experiments.

PR #316 (2020-12-12): [API] Data placement and streaming API rewrite
- [x] Fix the graph hierarchy issue (by reconstructing a call graph from the flattened graph)
- [x] Moved the code transformation logic to the end of the lowering process. Should fix #160 and fix #262 and fix #207 
- [x] Removed the automatically created on-chip buffer when moving data to/from the device. Users can choose from DMA or Stream mode for host-device communication (DMA is enabled by default) Should fix #235.
- [x] Constraints checking (e.g. one-read-one-write e.t.c)
- [x] Fixed the conflicts with `resue_at` (by adding an IR pass to fix  [... cut]