cornell-zhang/heterocl #160: Support `StreamExpr`  in reuse pattern detection

Hecmay, 2020-02-20:
For now `reuse_at` can only handle the reuse pattern found in Load expression. However, the Streaming IR pass replaces some potentially reusable Load expressions with StreamExpr expressions, making the `generate_reuse_buffer` IR pass crash with SegFault.

Example of generated Intel OpenCL code with streaming channel as followed. the Input image is replaced with `StreamExpr` in this case(i.e., `read_channel_intel(c_buf_4)`), applying `resue_at` on this input will result in Seg Fault.
```c++
__kernel void grad_weight_y() {
    float g_f[7];
    g_f[0] = 7.550000e-02f;
    g_f[1] = 1.330000e-01f;
    g_f[2] = 1.869000e-01f;
    g_f[3] = 2.903000e-01f;
    g_f[4] = 1.869000e-01f;
    g_f[5] = 1.330000e-01f;
    g_f[6] = 7.550000e-02f;
    for (int32_t y = 0; y < 430; ++y) {
      for (int32_t x = 0; x < 1024; ++x) {
        float reducer2;
        reducer2 = 0.000000e+00f;
        for (int32_t rdx = 0; rdx < 7; ++rdx) {
          reducer2 = ((read_channel_intel(c_buf_4) * g_f[rdx]) + reducer2);
        }
        grad_weight_y_y_filt[((x + (y * 1024)) + 3072)] = reducer2;
      }
    }
    #pragma ii 1
    for (int32_t buf_1 = 0; buf_1 < 1024; ++buf_1) {
      for (int32_t buf_0 = 0; buf_0 < 436; ++buf_0) {
        write_channel_intel(c_buf_5, grad_weight_y_y_filt[(buf_1 + (buf_0 * 1024))]);
      }
    }

```

zhangzhiru, 2020-02-21:
@Hecmay What is read_channel_intel? Note that we should not have target-specific primitive in our IR. This is supposed to be handled by different back-ends.

seanlatias, 2020-02-21:
I think he is showing the generated Intel OpenCL code, not the IR. For IR, we only have channel_read and channel_write. The final implementation is back-end dependent.

Hecmay, 2020-02-21:
`read_channel_intel` is the API function provided in Intel FPGA SDK for OpenCL to read data from a streaming channel. It corresponds to `StreamStmt` or `StreamExpr` IR node in HeteroCL IR. The IR representation is similar to the generated code.

zhangzhiru, 2020-02-21:
I see. I was confused by "StreamExpr (i.e., read_channel_intel)". These two are not supposed to be listed in parallel then.

Hecmay, 2021-09-20:
The issue is closed since it does not exist anymore. In the current implementation, s.to() only inserts some annotations into the IR instead of mutating the IR directly. The lowering function will replace the annotated Load IR node into StreamExpr node after the reuse buffer is inserted into the IR. In other words, the reuse pattern detection function only needs to handle Load IR nodes, and later passes will replace annotated Load IR nodes with StreamExpr nodes.

PR #316 (2020-12-12): [API] Data placement and streaming API rewrite
- [x] Fix the graph hierarchy issue (by reconstructing a call graph from the flattened graph)
- [x] Moved the code transformation logic to the end of the lowering process. Should fix #160 and fix #262 and fix #207 
- [x] Removed the automatically created on-chip buffer when moving data to/from the device. Users can choose from DMA or Stream mode for host-device communication (DMA is enabled by default) Should fix #235.
- [x] Constraints checking (e.g. one-read-one-write e.t.c)
- [x] Fixed the conflicts with `resue_at` (by adding an IR pass to fix  [... cut]
closed 2021-09-20 by Hecmay