cornell-zhang/heterocl #173: VTA does not compile on CircleCI

Hecmay, 2020-03-29:
The testing on CircleCI failed with the following error. It says the compiler cannot find the config file for VTA. However I can run through the script on our servers and everything works fine.
```shell
/usr/local/bin/python: can't open file 'vta/vta-hw/config/vta_config.py': [Errno 2] No such file or directory
-- Build VTA runtime with target: 
/usr/local/bin/python: can't open file 'vta/vta-hw/config/vta_config.py': [Errno 2] No such file or directory
-- Build with contrib.sort
-- Build with contrib.hybriddump
```

seanlatias, 2020-03-29:
Can you be more specific? Which test? Which PR I should look at?

Hecmay, 2020-03-29:
> Can you be more specific? Which test? Which PR I should look at?
 Sorry for that. I found all my PRs do not run through (e.g. In this PR #170, I only added a python file in `hlib/op` and then it failed the testing).

Hecmay, 2020-03-29:
> > Can you be more specific? Which test? Which PR I should look at?
> 
> Sorry for that. I found all my PRs do not run through (e.g. In this PR #170, I only added a python file in `hlib/op` and then it failed the testing).

I guess you can try re-compile the v0.3 branch on CircleCI to reproduce the error. It can be a lot more easier if we have an interactive shell for debugging.

Hecmay, 2020-03-29:
The issue magically disappears if we re-run the testing flow.

closed 2020-03-29 by Hecmay