cornell-zhang/heterocl #175: Problem building

payamn, 2020-04-09:
I followed the instruction from http://heterocl.csl.cornell.edu/doc/installation.html.
During the build process, I got the following error (I am using ubuntu 18):
 
```
/tmp/easy_install-Jc5guN/kiwisolver-1.2.0/.eggs/cppy-1.1.0-py2.7.egg/cppy/include/cppy/ptr.h:292:2: error: ‘Py_hash_t’ does not name a type
  Py_hash_t hash() const
  ^~~~~~~~~

In file included from py/kiwisolver.cpp:10:0:
py/types.h:33:12: error: ‘PyType_Spec’ does not name a type; did you mean ‘PyType_Type’?
     static PyType_Spec TypeObject_Spec;
            ^~~~~~~~~~~
            PyType_Type

py/types.h:47:12: error: ‘PyType_Spec’ does not name a type; did you mean ‘PyType_Type’?
     static PyType_Spec TypeObject_Spec;
            ^~~~~~~~~~~
            PyType_Type

py/types.h:66:12: error: ‘PyType_Spec’ does not name a type; did you mean ‘PyType_Type’?
     static PyType_Spec TypeObject_Spec;
            ^~~~~~~~~~~
            PyType_Type

py/types.h:85:12: error: ‘PyType_Spec’ does not name a type; did you mean ‘PyType_Type’?
     static PyType_Spec TypeObject_Spec;
            ^~~~~~~~~~~
            PyType_Type

py/types.h:104:12: error: ‘PyType_Spec’ does not name a type; did you mean ‘PyType_Type’?
     static PyType_Spec TypeObject_Spec;
            ^~~~~~~~~~~
            PyType_Type
py/types.h:122:12: error: ‘PyType_Spec’ does not name a type; did you mean ‘PyType_Type’?
     static PyType_Spec TypeObject_Spec;
            ^~~~~~~~~~~
            PyType_Type
py/kiwisolver.cpp:163:1: error: ‘PyModuleDef_Slot’ does not name a type; did you mean ‘PyModule_GetDict’?
 PyModuleDef_Slot kiwisolver_slots[] = {
 ^~~~~~~~~~~~~~~~
 PyModule_GetDict
py/kiwisolver.cpp:169:20: error: variable ‘{anonymous}::PyModuleDef {anonymous}::moduledef’ has initializer but incomplete type
 struct PyModuleDef moduledef = {
                    ^~~~~~~~~
```

seanlatias, 2020-04-10:
It seems like the error is caused by some other Python packages.  I don't think we are using the ``kiwisolver`` package as shown in your error message. I would suggest you use Conda to create a clean environment and use Python 3 instead of Python 2 to install HeteroCL. Please let me know if this solves the problem. Thank you!

payamn, 2020-05-03:
Thank you, that solved the problem.

closed 2020-05-03 by payamn