cornell-zhang/heterocl #176: Simulation failed using RTL IP integration

Hecmay, 2020-04-10:
Vivado HLS provides users with `blackbox` option when adding source files to the working project. Users can add a json file specifying the RTL kernel (i.e IP we want to integrate) and its corresponding C model (i.e. behavior level model).  

```tcl
addfile -blacbox rtl_model.json
```

The #170 is currently using the blackbox to connect HLS kernel and RTL IPs. This method is not stable (the SW simulation oftentimes halts forever). Also it has a lot of limitations, making it hard to be extended to other IP cores.

Instead of using blackbox, we should consider each IP core as a kernel function (instead of a sub-function being called inside the top-level function), and connect the kernel functions in the OpenCL host program.

Blaok, 2020-04-10:
> Instead of using blackbox, we should consider each IP core as a kernel function (instead of a sub-function being called inside the top-level function), and connect the kernel functions in the OpenCL host program.

Do you mean [kernel-to-kernel streaming](https://www.xilinx.com/html_docs/xilinx2019_2/vitis_doc/Chunk308472235.html#ariaid-title5)?

Hecmay, 2020-04-10:
@Blaok. Yes, that's what I mean. We can create the k2k streaming channels, and all the kernel functions (either IPs or HLS functions) are enqueued into the command queue.

zhangzhiru, 2020-04-10:
Why is OpenCL host program required to support kernel-to-kernel streaming?

Blaok, 2020-04-10:
@Hecmay Good to know. SODA actually has some preliminary support for k2k streaming. I'll keep you posted when I have a fully working version ready for integration.

Hecmay, 2020-04-10:
> Why is OpenCL host program required to support kernel-to-kernel streaming?

The streaming channels between kernels are not declared in OpenCL host program. It's similar to `autorun` attribute in AOCL where the host program only enqueues the kernels, and data is streamed between those kernels.

zhangzhiru, 2020-04-10:
That's what I thought. So this is 
>> connect the kernel functions in the OpenCL host program.

seanlatias, 2020-04-10:
> ... and connect the kernel functions in the OpenCL host program.

So this part is misleading in your first post.

Hecmay, 2020-04-11:
> > ... and connect the kernel functions in the OpenCL host program.
> 
> So this part is misleading in your first post.

That's true. It's a bit misleading. Removed.