cornell-zhang/heterocl #181: Installation problems when using non-default tools

chhzh123, 2020-04-26:
The [installation page](http://heterocl.csl.cornell.edu/doc/installation.html) seems to miss lots of details. Though it suggests users set their own CMake and LLVM version, after I change `LLVM_CONFIG`, `HCL_PKG_LLVM`, and `HCL_PKG_LLVM_BIN` in `Makefile.config`, the build system still downloads LLVM 6.0 from the Internet by default. The reason comes from the current Makefile that the prerequisite of `build-tvm` is `build-pkgs`, which is no need for those who have installed LLVM or CMake.

Moreover, it seems HeteroCL uses a rather old TVM that does not support the new version of LLVM. There are some APIs changed for LLVM 9.0 or above, and it would be better if HeteroCL could consider different versions of LLVM for the codegen, e.g. the [SaveToFile](https://github.com/apache/incubator-tvm/blob/841725cc5858a196297bbd316a589afb2f87ab18/src/target/llvm/llvm_module.cc#L84) function in TVM.

I am also misled that the `make` instruction only builds the libraries, and users can install the built HeteroCL with their favorite python. However, I found it revoked python2 in my Linux, which is not I expected. Since `python` refers to Python 2 while `python3` refers to the correct Python 3 version in some Linux systems, I suggest separate this part for the default installation. Otherwise, provide a `$(PYTHON)` variable for users to change their python versions.

To sum up, the installation guide needs to be enhanced, and supports for newer versions of LLVM should be added.
After deleting the `build-pkgs` requisite, maybe something like
```bash
make build-tvm
export HCL_HOME=/path/to/heterocl
export PYTHONPATH=$HCL_HOME/python:$HCL_HOME/hlib/python:${PYTHONPATH}
```
would be preferable.

zhangzhiru, 2020-04-26:
@chhzh123 do you plan to submit a PR that fixes some of these issues.

Note that we are not too interested in supporting newer version of LLVM for now (as it's only useful for CPU sim right now). Regardless, our installation guide should be clear about the version we recommend; plus the build system should not re-download and rebuild LLVM even user has already set the local path.

chhzh123, 2020-04-27:
Well, then leave it without changes.

chhzh123, 2020-04-27:
> @chhzh123 do you plan to submit a PR that fixes some of these issues?
>
> Note that we are not too interested in supporting newer version of LLVM for now (as it's only useful for CPU sim right now). [...]

Well, then leave it without changes.

zhangzhiru, 2020-04-27:
You mentioned that the build system downloads LLVM 6 despite of the changes in Makefile.config. Have you fixed that issue?

chhzh123, 2020-04-27:
Yes, I found that results from the original configuration file that only recognizes LLVM 7.0 and below. I have created a PR to fix it.

closed 2020-04-29 by chhzh123