cornell-zhang/heterocl #190: " python hlib/tests/test_extern_ip.py " cann't work.

YongLeeee, 2020-05-04:
Working on the v0.3 branch with _python3.6_.

Traceback (most recent call last):
  File "hlib/tests/test_extern_ip.py", line 4, in <module>
    import hlib
  File "/home/yl924/git/heterocl/hlib/python/hlib/__init__.py", line 3, in <module>
    from . import frontend
  File "/home/yl924/git/heterocl/hlib/python/hlib/frontend/__init__.py", line 1, in <module>
    from . import relay_parser
  File "/home/yl924/git/heterocl/hlib/python/hlib/frontend/relay_parser.py", line 1, in <module>
    import tvm
ModuleNotFoundError: No module named 'tvm'

seanlatias, 2020-05-04:
For now, you can just comment out the `from . import relay_parser`. Otherwise, you need to install tvm.

YongLeeee, 2020-05-04:
It's the solution.

closed 2020-05-04 by YongLeeee