cornell-zhang/heterocl #194: Vivado HLS compilation fails for v0.3 with new function signature

chhzh123, 2020-05-05:
The new v0.3 generates flat arrays and passes in pointers in default. Vivado HLS cannot infer the size of the arrays, leading to a compile error.

I still use [Tutorial: Back-end Support](http://heterocl.csl.cornell.edu/doc/tutorials/tutorial_08_backend.html#vivado-hls-c-code-generation) to show different generated code here.
```cpp
// v0.2
void default_function(ap_int<32> A[10][10], ap_uint<4> B[8][8]) {
#pragma HLS array_partition variable=A complete dim=0
  for (ap_int<32> y = 0; y < 8; ++y) {
    for (ap_int<32> x = 0; x < 8; ++x) {
    #pragma HLS pipeline
      B[y][x] = ((ap_uint<4>)(((ap_int<33>)A[y][x]) + ((ap_int<33>)A[(y + 2)][(x + 2)])));
    }
  }
}

// v0.3
void default_function(ap_int<32>* A, ap_uint<4>* B) { // pass in pointers
                  #pragma HLS array_partition variable=A complete dim=0
  for (ap_int<32> y = 0; y < 8; ++y) {
    for (ap_int<32> x = 0; x < 8; ++x) {
    #pragma HLS pipeline
      B[(x + (y * 8))] = ((ap_uint<4>)(((ap_int<33>)A[(x + (y * 10))]) + ((ap_int<33>)A[((x + (y * 10)) + 22)])));
    }
  }
}
```

The new function signature results in the following error.
```cpp
ERROR: [SYNCHK 200-61] ./simple_add.cpp:12: unsupported memory access on variable 'A.V' which is (or contains) an array with unknown size at compile time.
```

seanlatias, 2020-05-05:
@Hecmay  can you take a look at this?

Hecmay, 2020-05-05:
Sure. The recommended way to generate code is to use `hcl.build(s, platform)` instead of passing in strings. We should also update the document for v0.3 some time.

seanlatias, 2020-05-05:
So what is wrong here?

seanlatias, 2020-05-05:
The string one is no longer supported?

Hecmay, 2020-05-05:
It's still supported. I mean if you use the old interface, you need to dump the returned string into a file, manually write the TCL script to run HLS, and run viavdo_hls. Just more efforts, nothing else.

zhangzhiru, 2020-05-05:
This sounds like another case where we should prompt error messages.

seanlatias, 2020-05-05:
> It's still supported. I mean if you use the old interface, you need to dump the returned string into a file, manually write the TCL script to run HLS, and run viavdo_hls. Just more efforts, nothing else.

Not sure if I understand. But the problem here is that it seems like the pointer arguments cause the errors. It doesn't matter if we are using strings or hcl.platform.

Hecmay, 2020-05-05:
Sure. That's the problem. I will fix the issue.

seanlatias, 2020-05-06:
@chhzh123 Please check the fix and see if that solves the problem.

chhzh123, 2020-05-06:
Yes, it looks fine. But it means in v0.3, users need to manually flatten their tensors to 1D arrays before calling the function? As the signature depicts.

```cpp
void default_function(ap_int<32> A[10*10], ap_uint<4> B[8*8]);
```

Hecmay, 2020-05-06:
The users do not need to manually flatten the array index. The reason of the index flattening is that sometimes Vivado HLS will throw out errors when accessing an array with more than 3 index dimensions. Not sure whether this issue has been fixed in the newer version. I will check it out.

Hecmay, 2020-05-06:
Similar issue as what I got: https://forums.xilinx.com/t5/High-Level-Synthesis-HLS/Can-HLS-handle-3-dimension-arrays/td-p/799164.

chhzh123, 2020-06-13:
> The users do not need to manually flatten the array index. The reason of (automatic) index flattening is that sometimes Vivado HLS will throw out errors when accessing an array with more than 3 index dimensions. Not sure whether this issue has been fixed in the newer version. I will check it out.

#206 changes back to multi-dimensional arrays. I have checked the functionality on Vivado HLS (v2018), and N-D arrays can be correctly synthesized.

zhangzhiru, 2020-06-13:
@chhzh123 can you help add a test case under tests/issues/194?

chhzh123, 2020-06-14:
If we want to check the correctness of synthesis, we need to fix the mode of DUT first.

chhzh123, 2020-06-14:
I'll discuss with @Hecmay and bring up a solution for DUT. Then add csyn tests for this issue and the programs in tutorials like #233 .

Hecmay, 2021-09-23:
This should have been fixed in https://github.com/cornell-zhang/heterocl/pull/418. 
Test case added in tests/issues/test_issue_194.py. Please reopen the issue if the error still exists.

PR #195 (2020-05-06): [Backend] Fix Vivado HLS CodeGen for data type casting and function signature 
Should fix #194 and #193
PR #316 (2020-12-12): [API] Data placement and streaming API rewrite
- [x] Fix the graph hierarchy issue (by reconstructing a call graph from the flattened graph)
- [x] Moved the code transformation logic to the end of the lowering process. Should fix #160 and fix #262 and fix #207 
- [x] Removed the automatically created on-chip buffer when moving data to/from the device. Users can choose from DMA or Stream mode for host-device communication (DMA is enabled by default) Should fix #235.
- [x] Constraints checking (e.g. one-read-one-write e.t.c)
- [x] Fixed the conflicts with `resue_at` (by adding an IR pass to fix  [... cut]
closed 2021-09-23 by Hecmay