cornell-zhang/heterocl #207: Test_runtime_build failed using `hcl.platform`

chhzh123, 2020-05-13:
There are no documents about the usage of `hcl.platform`, and I can only guess from the provided examples. However, when I run the `test_vivado_hls` function in [test_runtime_build.py](https://github.com/cornell-zhang/heterocl/blob/39cd79c09357450cd6873686fe740acaddbef6d7/tests/test_runtime_build.py#L45), I received the following error.
```
Traceback (most recent call last):
  File "test_runtime_build.py", line 41, in <module>
    test_vivado_hls()
  File "test_runtime_build.py", line 30, in test_vivado_hls
    f(hcl_A, hcl_B)
  File "/home/chz/heterocl/python/heterocl/tvm/_ffi/function.py", line 128, in __call__
    return f(*args)
  File "/home/chz/heterocl/python/heterocl/tvm/_ffi/_ctypes/function.py", line 183, in __call__
    ctypes.byref(ret_val), ctypes.byref(ret_tcode)))
  File "/home/chz/heterocl/python/heterocl/tvm/_ffi/base.py", line 66, in check_call
    raise TVMError(py_str(_LIB.TVMGetLastError()))
heterocl.tvm._ffi.base.TVMError: [17:29:09] src/codegen/build_util.cc:376: Check failed: test_file.find("test(") != std::string::npos cannot find top function
```

`USE_VIVADO_HLS` is set, and `vivado_hls` is also in system path.

What I have done is to change the platform from `aws_f1` to `zc706` and comment the [streaming commands](https://github.com/cornell-zhang/heterocl/blob/master/tests/test_runtime_build.py#L59-#L60).

I think this probably is not a bug, since the program seems to pass the test on the cloud. But I wonder how I can correctly do simulation and synthesis in HeteroCL. Thanks!

Hecmay, 2020-05-13:
That's a problem from the header generation logic. For Vivado HLS backend, HeteroCL analyzes the generated kernel code, and try to locate the top-level function in the code string (i.e. the `test` function here). 

Since `.to` is not used in this case, the whole program will run on the host CPU. As a result, there is no top-level function found in the generated code, and thus the error occurred. A simple workaround is to add `.to` back to the program.  I will update the header generation logic in #206.

Hecmay, 2020-05-13:
Another issue caused by the header generation logic is that the pragmas are hardcoded for VHLS SDSoC mode (this is a quick and dirty solution to make it work... will fix it very soon) https://github.com/cornell-zhang/heterocl/blob/master/tvm/src/codegen/build_util.cc#L382-L404

zhangzhiru, 2020-05-13:
@Hecmay when do we plan to add the user document on the usage of the hcl.platform?

Hecmay, 2020-05-13:
I will discuss with Sean. We can add a short tutorial to showcase how to use the platform feature.

chhzh123, 2020-05-13:
> A simple workaround is to add `.to` back to the program.

Thanks. This method works for software simulation, but how can I run synthesis?

I found there is a condition of `platform == "vivado"` in [`build_module.py`](https://github.com/cornell-zhang/heterocl/blob/master/python/heterocl/tvm/build_module.py#L47-L49), so I change the target of `test_vivado_hls` to `target.config(compile="vivado", mode="hw_sim")`, but it leads to errors again.
```
Traceback (most recent call last):
  File "test_runtime_build.py", line 42, in <module>
    test_vivado_hls()
  File "test_runtime_build.py", line 24, in test_vivado_hls
    f = hcl.build(s, target)
  File "/home/chz/heterocl/python/heterocl/api.py", line 318, in build
    return _build(schedule.sch, new_inputs, target=target, name=name, stmt=stmt)
  File "/home/chz/heterocl/python/heterocl/tvm/build_module.py", line 768, in build
    return build_fpga_kernel(sch, args, target, name=name)
  File "/home/chz/heterocl/python/heterocl/tvm/build_module.py", line 716, in build_fpga_kernel
    return builder(fdevice, keys, vals)
  File "/home/chz/heterocl/python/heterocl/tvm/_ffi/function.py", line 280, in my_api_func
    return flocal(*args)
  File "/home/chz/heterocl/python/heterocl/tvm/_ffi/_ctypes/function.py", line 183, in __call__
    ctypes.byref(ret_val), ctypes.byref(ret_tcode)))
  File "/home/chz/heterocl/python/heterocl/tvm/_ffi/base.py", line 66, in check_call
    raise TVMError(py_str(_LIB.TVMGetLastError()))
heterocl.tvm._ffi.base.TVMError: [23:55:29] include/tvm/./packed_func_ext.h:101: Check failed: NodeTypeChecker<TNodeRef>::Check(sptr.get()) Expected type array<Expr> but get Array
```

Hecmay, 2020-05-13:
I have not checked the Vivado synthesis flow yet. Will check it now.

chhzh123, 2020-06-13:
> Thanks. This method works for software simulation, but how can I run synthesis?
>
> I found there is a condition of `platform == "vivado"` in [...]

Use `target.config(compile="vivado_hls",mode="csyn")` to run synthesis. This mode has been added in #215 .

chhzh123, 2020-06-13:
> Since `.to` is not used in this case, the whole program will run on the host CPU. As a result, there is no top-level function found in the generated code, and thus the error occurred. A simple workaround is to add `.to` back to the program. I will update the header generation logic in #206.

#206 does not fix this issue. When no `.to` is specified, it causes errors.

Hecmay, 2020-06-13:
I will add a primitive as a simple prototyping for compute placement. And instead of using the stream channels inside the kernel function, I will add a new IR pass to convert the on-chip buffer accessing to off-chip buffer accessing. This should address your concern. Will work on this and keep you posted.

Hecmay, 2021-09-23:
This should have been fixed in https://github.com/cornell-zhang/heterocl/pull/418. 
Test case added in tests/issues/test_issue_207.py. Please reopen the issue if the error still exists.

PR #316 (2020-12-12): [API] Data placement and streaming API rewrite
- [x] Fix the graph hierarchy issue (by reconstructing a call graph from the flattened graph)
- [x] Moved the code transformation logic to the end of the lowering process. Should fix #160 and fix #262 and fix #207 
- [x] Removed the automatically created on-chip buffer when moving data to/from the device. Users can choose from DMA or Stream mode for host-device communication (DMA is enabled by default) Should fix #235.
- [x] Constraints checking (e.g. one-read-one-write e.t.c)
- [x] Fixed the conflicts with `resue_at` (by adding an IR pass to fix  [... cut]
closed 2021-09-23 by Hecmay