cornell-zhang/heterocl #208: Unable to run "make -j8 or make -j4" command in the downloaded HeteroCL directory.

tyson98, 2020-05-13:
This is Jayesh Sharma, from IIT Kharagpur. I am facing some trouble in installing the packages of HeteroCL. After downloading the packages as I make/build them using the "make -j8" command I incur the following errors. 

jayesh@jayesh:~/heterocl$ make -j8
make -C pkgs
make[1]: Entering directory '/home/jayesh/heterocl/pkgs'
make[2]: Entering directory '/home/jayesh/heterocl/pkgs/cmake'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/jayesh/heterocl/pkgs/cmake'
make[2]: Entering directory '/home/jayesh/heterocl/pkgs/llvm'
cd /home/jayesh/heterocl/build/pkgs/llvm/build; \
/home/jayesh/heterocl/build/pkgs/cmake/build/cmake/bin/cmake -G "Unix Makefiles" -DCMAKE_INSTALL_PREFIX=/home/jayesh/heterocl/build/pkgs/install -DCMAKE_BUILD_TYPE=Release /home/jayesh/heterocl/build/pkgs/llvm/src
-- Native target architecture is X86
-- Threads enabled.
-- Doxygen disabled.
-- Go bindings disabled.
-- Could NOT find OCaml (missing: OCAMLFIND OCAML_VERSION OCAML_STDLIB_PATH) 
-- Could NOT find OCaml (missing: OCAMLFIND OCAML_VERSION OCAML_STDLIB_PATH) 
-- OCaml bindings disabled.
-- Could NOT find Python module pygments
-- Could NOT find Python module pygments.lexers.c_cpp
-- Could NOT find Python module yaml
-- LLVM host triple: x86_64-unknown-linux-gnu
-- LLVM default target triple: x86_64-unknown-linux-gnu
-- Building with -fPIC
-- Constructing LLVMBuild project information
-- Linker detection: GNU ld

Hence, I am unable to run the tutorials in the Jupyter notebook. 

The specification of my machine is as follows:
Windows 10 machine, and running Ubuntu 20.04 as a virtual machine using the Virtual Box of Oracle. 
I have tried to remove some of the errors by installing OCaml, python-payments, yalm. But still, it is unable to find them.
I believe ubuntu 20.04 is not properly able to install all the packages for python3. Though it shows it has python3.

seanlatias, 2020-05-13:
What's the final error message?

tyson98, 2020-05-14:
Sir, I am attaching the entire message log of my terminal where I execute
the following command.
ᐧ

jayesh@jayesh:~/heterocl$ make -j8
make -C pkgs
make[1]: Entering directory '/home/jayesh/heterocl/pkgs'
make[2]: Entering directory '/home/jayesh/heterocl/pkgs/cmake'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/jayesh/heterocl/pkgs/cmake'
make[2]: Entering directory '/home/jayesh/heterocl/pkgs/llvm'
cd /home/jayesh/heterocl/build/pkgs/llvm/build; \
/home/jayesh/heterocl/build/pkgs/cmake/build/cmake/bin/cmake -G "Unix Makefiles" -DCMAKE_INSTALL_PREFIX=/home/jayesh/heterocl/build/pkgs/install -DCMAKE_BUILD_TYPE=Release /home/jayesh/heterocl/build/pkgs/llvm/src
-- Native target architecture is X86
-- Threads enabled.
-- Doxygen disabled.
-- Go bindings disabled.
-- Could NOT find OCaml (missing: OCAMLFIND OCAML_VERSION OCAML_STDLIB_PATH) 
-- Could NOT find OCaml (missing: OCAMLFIND OCAML_VERSION OCAML_STDLIB_PATH) 
-- OCaml bindings disabled.
-- Could NOT find Python module pygments
-- Could NOT find Python module pygments.lexers.c_cpp
-- Could NOT find Python module yaml
-- LLVM host triple: x86_64-unknown-linux-gnu
-- LLVM default target triple: x86_64-unknown-linux-gnu
-- Building with -fPIC
-- Constructing LLVMBuild project information
-- Linker detection: GNU ld
-- Targeting AArch64
-- Targeting AMDGPU
-- Targeting ARM
-- Targeting BPF
-- Targeting Hexagon
-- Targeting Lanai
-- Targeting Mips
-- Targeting MSP430
-- Targeting NVPTX
-- Targeting PowerPC
-- Targeting Sparc
-- Targeting SystemZ
-- Targeting X86
-- Targeting XCore
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jayesh/heterocl/build/pkgs/llvm/build
cd /home/jayesh/heterocl/build/pkgs/llvm/build; \
make
make[3]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[4]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
[  0%] Built target LLVMHello_exports
[  0%] Built target llvm_vcsrevision_h
[  1%] Built target LLVMTableGen
[  1%] Built target LLVMDemangle
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
[... 864 lines cut ...]
make[4]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[3]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[2]: Leaving directory '/home/jayesh/heterocl/pkgs/llvm'
make[1]: Leaving directory '/home/jayesh/heterocl/pkgs'
make -C tvm
make[1]: Entering directory '/home/jayesh/heterocl/tvm'
cd HalideIR; make lib/libHalideIR.a DMLC_CORE_PATH=../dmlc-core; cd /home/jayesh/heterocl/tvm
make[2]: Entering directory '/home/jayesh/heterocl/tvm/HalideIR'
make[2]: 'lib/libHalideIR.a' is up to date.
make[2]: Leaving directory '/home/jayesh/heterocl/tvm/HalideIR'
make[1]: Leaving directory '/home/jayesh/heterocl/tvm'
cd python; \
python setup.py install --user; \
  cd ../hlib/python; \
python setup.py install --user;
Traceback (most recent call last):
  File "setup.py", line 1, in <module>
    from setuptools import setup, find_packages
ImportError: No module named setuptools
Traceback (most recent call last):
  File "setup.py", line 1, in <module>
    from setuptools import setup, find_packages
ImportError: No module named setuptools
make: *** [Makefile:14: build-hcl] Error 1
jayesh@jayesh:~/heterocl$

tyson98, 2020-05-14:
Also, I have tried debugging the "No module named setuptools" error following the commands given in this link: https://stackoverflow.com/questions/14426491/python-3-importerror-no-module-named-setuptools

Hecmay, 2020-05-15:
@tyson98 Can you try to install heterocl with conda (you need to install conda first):
```shell
conda install -c cornell-zhang heterocl
```

ted-xie, 2020-05-16:
@tyson98 Try `python -c "import setuptools"`. If that fails then there is a problem with your system environment.

tyson98, 2020-05-16:
@ted-xie the following command worked. I am able to run HeteroCL on Jupyter notebook now. 
Thank you @ted-xie, @Hecmay and @seanlatias.

closed 2020-05-25 by Hecmay

tyson98, 2020-06-07:
Hello Everyone, I am again facing the same issue of running HeteroCL on Juptyer Notebook. It is unable to find attributes such as init, build, etc. 
I have recently tried to install Miniconda in my machine, before that it was working fine. 
I have deleted and re-installed the entire package by both the ways mentioned. Still, it's not working. 

The message I am getting as I run the "make -j4" is as follows: 

jayesh@jayesh:~/heterocl$ make -j4
make -C pkgs
make[1]: Entering directory '/home/jayesh/heterocl/pkgs'
make[2]: Entering directory '/home/jayesh/heterocl/pkgs/cmake'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/jayesh/heterocl/pkgs/cmake'
make[2]: Entering directory '/home/jayesh/heterocl/pkgs/llvm'
cd /home/jayesh/heterocl/build/pkgs/llvm/build; \
/home/jayesh/heterocl/build/pkgs/cmake/build/cmake/bin/cmake -G "Unix Makefiles" -DCMAKE_INSTALL_PREFIX=/home/jayesh/heterocl/build/pkgs/install -DCMAKE_BUILD_TYPE=Release /home/jayesh/heterocl/build/pkgs/llvm/src
-- Native target architecture is X86
-- Threads enabled.
-- Doxygen disabled.
-- Go bindings disabled.
-- Could NOT find OCaml (missing: OCAMLFIND OCAML_VERSION OCAML_STDLIB_PATH) 
-- Could NOT find OCaml (missing: OCAMLFIND OCAML_VERSION OCAML_STDLIB_PATH) 
-- OCaml bindings disabled.
-- Could NOT find Python module pygments
-- Could NOT find Python module pygments.lexers.c_cpp
-- Could NOT find Python module yaml
-- LLVM host triple: x86_64-unknown-linux-gnu
-- LLVM default target triple: x86_64-unknown-linux-gnu
-- Building with -fPIC
-- Constructing LLVMBuild project information
-- Linker detection: GNU ld
-- Targeting AArch64
-- Targeting AMDGPU
-- Targeting ARM
-- Targeting BPF
-- Targeting Hexagon
-- Targeting Lanai
-- Targeting Mips
-- Targeting MSP430
-- Targeting NVPTX
-- Targeting PowerPC
-- Targeting Sparc
-- Targeting SystemZ
-- Targeting X86
-- Targeting XCore
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jayesh/heterocl/build/pkgs/llvm/build
cd /home/jayesh/heterocl/build/pkgs/llvm/build; \
make
make[3]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[4]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
[  0%] Built target LLVMDemangle
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
[  1%] Built target LLVMTableGen
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
[  1%] Built target LLVMBinaryFormat
make[5]: Entering directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
[  1%] Built target LLVMHello_exports
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
[  1%] Built target llvm_vcsrevision_h
make[5]: Leaving directory '/home/jayesh/heterocl/build/pkgs/llvm/build'
[... 707 lines cut ...]
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: | 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                          

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - heterocl -> python[version='>=3.8,<3.9.0a0']

Your python: python=3.7

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.

jayesh@jayesh:~$

seanlatias, 2020-06-09:
As you can see, you have some missing packages.

``ImportError: No module named setuptools``

Hecmay, 2020-06-09:
Like Sean mentioned. Solution is to run `pip install setuptools`. 

I f you wan to use conda to install HeteroCL, you should create a conda env first.

tyson98, 2020-06-09:
I have tried the following commands, and still getting the same error. 

**jayesh@jayesh:**~/heterocl$ pip install setuptools
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: setuptools in /home/jayesh/.local/lib/python3.8/site-packages (47.1.1)

**jayesh@jayesh:**~/heterocl$ sudo apt-get install python3-setuptools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-setuptools is already the newest version (45.2.0-1).
0 upgraded, 0 newly installed, 0 to remove and 127 not upgraded.

**jayesh@jayesh:**~/heterocl$ pip install -U pip setuptools 
Defaulting to user installation because normal site-packages is not writeable
Requirement already up-to-date: pip in /home/jayesh/.local/lib/python3.8/site-packages (20.1.1)
Requirement already up-to-date: setuptools in /home/jayesh/.local/lib/python3.8/site-packages (47.1.1)

**jayesh@jayesh:**~/heterocl$ pip install --upgrade setuptools
Defaulting to user installation because normal site-packages is not writeable
Requirement already up-to-date: setuptools in /home/jayesh/.local/lib/python3.8/site-packages (47.1.1)

seanlatias, 2020-06-09:
Can you just create a conda virtual environment and install everything in the environment? It seems that there are some path finding issues and those are not related to HeteroCL.