cornell-zhang/heterocl #213: SODA backend cannot generate XHLS code

Hecmay, 2020-05-19:
Minimal test case to reproduce error:
```python
def test_stencil_multi_stencil():
    A = hcl.placeholder((10, 10), "A")
    def kernel(A):
        B = hcl.compute((10, 8), lambda y, x: A[y, x] + A[y, x+1] + A[y, x+2], "B")
        C = hcl.compute((8, 8), lambda y, x: B[y, x] + B[y+1, x] + B[y+2, x], "C")
    s = hcl.create_schedule(A, kernel)
    s[kernel.B].stencil(burst_width=256, unroll_factor=4)
    s[kernel.C].stencil(burst_width=128, unroll_factor=8)
    print(hcl.build(s, target='soda_xhls'))
```

The generated IR is correct, but SODAC failed to generate XHLS code from the IR
```shell
[21:54:06] src/codegen/build_soda.cc:90: ERROR:sodac:130: None:10:1: error: Expected '*' or '/' or '%' or '+' or '-' or '<=' or '>=' or '<' or '>' or '==' or '!=' or BinaryAndOp or XorOp or BinaryOrOp or LogicAndOp or LogicOrOp or 'local' or 'output' or 'param' at position (10, 1) => '(2, 0)))) *'.
```

Blaok, 2020-05-19:
The SODA backend is not aware of the `Stencil` IR node. I'll work on it.

Blaok, 2020-05-20:
Question about the semantics: the above example contains two separate `Stencil` IR nodes. What should I return from `hcl.build`? HLS code concatenated?

Hecmay, 2020-05-20:
I think so. It is supposed to return the concatenated HLSC kernel code, or use the similar semantics for Stencil Nodes in VHLS CodeGen (e.g. return the function calls only and save the definition inside a separate file)? @seanlatias What do you think?

Hecmay, 2020-05-22:
And also a follow-up to the issue. The program above cannot run through the Vivado HLS CodeGen. The error message is as followed:
```shell
[21:16:58] src/codegen/build_soda.cc:90: Traceback (most recent call last):
  File "/work/shared/users/phd/sx233/heterocl/soda/src/sodac", line 139, in <module>
    main()
  File "/work/shared/users/phd/sx233/heterocl/soda/src/sodac", line 127, in main
    xocl.print_code(stencil, args)
  File "/work/shared/users/phd/sx233/heterocl/soda/src/soda/codegen/xilinx/opencl.py", line 36, in print_code
    kernel.print_code(stencil, tmp)
  File "/work/shared/users/phd/sx233/heterocl/soda/src/soda/codegen/xilinx/hls_kernel.py", line 173, in print_code
    burst_width=stencil.burst_width)
  File "/work/shared/users/phd/sx233/heterocl/soda/src/soda/codegen/xilinx/hls_kernel.py", line 282, in _print_module_definition
    raise util.InternalError('cannot process such a burst yet')
haoda.util.InternalError: cannot process such a burst yet
```

Another question about how we can connect those stencil kernels with FIFOs: each stencil node is mapped to a sub-kernel function with m_axi interface, and there are corresponding `BurstRead` and `BurstWrite`functions to stream the data into/from the streaming channels. Can SODAC generate HLSC code fr Stencil node without the AXIS master interfaces, and replace it with the streaming channels?

Blaok, 2020-05-22:
The error is supposed to go away with the `Stencil` IR node being used.

Regarding integration, replacing `m_axi` with a streaming interface is certainly doable.

1. I'll add a new backend to SODA that generates HLS code with `axis` interface instead of `m_axi`.
2. The `axis` interface will not support `TLAST` to remove some resource overhead. This requires that whoever reading from the interface must know when to stop by itself (which is likely the case because it knows the loop trip count).
3. The HLS code is only supposed to be used for software simulation. For hardware simulation/bitstream generation, xo files must be exported directly from SODA to guarantee full pipelining.
4. I'll update the SODA documentation with details about the data layout.

I probably won't have bandwidth to actually implement those until the ICCAD deadline. I'll keep you posted.

Hecmay, 2020-05-23:
Thanks! It's totally fine. No rush.

I will also update the IR integration part to support importing RTL modules. We can discuss later after the deadline.

PR #225 (2020-06-01): [Backend][SODA] use Stencil IR node
Refactor the SODA backend.

Fixes #213.
closed 2020-06-01 by Blaok