cornell-zhang/heterocl #214: Stream port has invalid interface mode 's_axilite' for Vivado HLS

chhzh123, 2020-05-20:
Following the guidance in #207 , I can do simulation for the `test_vivado_hls` function in [`test_runtime_build.py`](https://github.com/cornell-zhang/heterocl/blob/39cd79c09357450cd6873686fe740acaddbef6d7/tests/test_runtime_build.py#L45). However, the program still cannot pass the HLS flow.

The generated kernel code is listed below.
```cpp
  void test(hls::stream<bit32>& B_channel, hls::stream<bit32>& C_channel) {
  #pragma HLS INTERFACE axis port=B_channel depth=1 offset=slave bundle=gmem0
  #pragma HLS INTERFACE axis port=C_channel depth=1 offset=slave bundle=gmem1
  #pragma HLS INTERFACE s_axilite port=B_channel bundle=control
  #pragma HLS INTERFACE s_axilite port=C_channel bundle=control
  #pragma HLS INTERFACE s_axilite port=return bundle=control
    bit32 B[320];
    for (bit32 B0 = 0; B0 < 10; ++B0) {
      for (bit32 B1 = 0; B1 < 32; ++B1) {
        B[(B1 + (B0 * 32))] = B_channel.read();
      }
    }
    bit32 C[320];
    for (bit32 args = 0; args < 10; ++args) {
      for (bit32 args0 = 0; args0 < 32; ++args0) {
        C[(args0 + (args * 32))] = (B[(args0 + (args * 32))] + 1);
      }
    }
    for (bit32 C0 = 0; C0 < 10; ++C0) {
      for (bit32 C1 = 0; C1 < 32; ++C1) {
        C_channel.write(C[(C1 + (C0 * 32))]);
      }
    }
  }
```

And Vivado HLS (v2018) failed to synthesize, giving the following error.
```
WARNING: [XFORM 203-803] Dropped interface mode 'axis' on 'B_channel.V.V' (kernel.cpp:9) as it is incompatible with its interface mode 's_axilite'.
WARNING: [XFORM 203-803] Dropped interface mode 'axis' on 'C_channel.V.V' (kernel.cpp:9) as it is incompatible with its interface mode 's_axilite'.
ERROR: [XFORM 203-801] Stream port 'B_channel.V.V' (kernel.cpp:9) has invalid interface mode 's_axilite' (kernel.cpp:12:1). Stream port only supports ap_hs, ap_fifo and axis modes.
ERROR: [XFORM 203-801] Stream port 'C_channel.V.V' (kernel.cpp:9) has invalid interface mode 's_axilite' (kernel.cpp:13:1). Stream port only supports ap_hs, ap_fifo and axis modes.
ERROR: [HLS 200-70] Pre-synthesis failed.
```

Moreover, the [`run.tcl` script](https://github.com/cornell-zhang/heterocl/blob/e322a080e755f6bc1178965c61c30ef7afe524e5/tvm/src/template/vivado/run.tcl#L13) is out-of-date, where the top function has not been changed to `test`.

Hecmay, 2020-05-20:
Thanks! There should not be any `s_axilite` interface mode specified for a streaming port. This bug is fixed in #206. This PR will be merged after I add the double buffer and p2p streaming support from FPGA to NVMe. 

The coverage of the test cases is not very complete.  For the testing function you mentioned, I only tested the case for C-simulation: https://github.com/cornell-zhang/heterocl/blob/master/tvm/src/template/vivado/Makefile#L19

chhzh123, 2020-05-20:
> There should not be any `s_axilite` interface mode specified for a streaming port. This bug is fixed in #206.

OK, thanks, I'll have a check.

chhzh123, 2020-05-23:
Fixed in #215 .

closed 2020-05-23 by chhzh123