cornell-zhang/heterocl #216: SegFault when testing hlib keras

Hecmay, 2020-05-22:
```shell
============================= test session starts ==============================
platform linux -- Python 3.6.10, pytest-5.4.2, py-1.8.1, pluggy-0.13.1
rootdir: /home/circleci/project
collecting ... collecting 1 item                                                              collected 78 items                                                             

hlib/tests/test_batch_norm.py .                                          [  1%]
hlib/tests/test_bias_add.py .                                            [  2%]
hlib/tests/test_broadcast.py ..........                                  [ 15%]
hlib/tests/test_elemwise.py ...............                              [ 34%]
hlib/tests/test_extern_ip.py .                                           [ 35%]
hlib/tests/test_flatten.py .                                             [ 37%]
hlib/tests/test_keras.py ............Fatal Python error: Segmentation fault

Current thread 0x00007f49c7979740 (most recent call first):
  File "/home/circleci/.local/lib/python3.6/site-packages/tvm-0.7.dev1-py3.6-linux-x86_64.egg/tvm/_ffi/_ctypes/packed_func.py", line 212 in __call__
  File "/home/circleci/.local/lib/python3.6/site-packages/tvm-0.7.dev1-py3.6-linux-x86_64.egg/tvm/runtime/object.py", line 53 in __getattr__
  File "/home/circleci/project/hlib/python/hlib/op/nn.py", line 557 in split
  File "/home/circleci/project/hlib/python/hlib/frontend/relay_parser.py", line 276 in gen_code
  File "/home/circleci/project/hlib/python/hlib/frontend/relay_parser.py", line 314 in func
  File "/home/circleci/project/python/heterocl/api.py", line 207 in create_schedule
  File "/home/circleci/project/hlib/python/hlib/frontend/relay_parser.py", line 685 in get_relay_model
  File "/home/circleci/project/hlib/tests/test_keras.py", line 31 in get_hcl_output
  File "/home/circleci/project/hlib/tests/test_keras.py", line 52 in verify_keras_frontend
  File "/home/circleci/project/hlib/tests/test_keras.py", line 325 in test_rnn
  File "/home/circleci/.local/lib/python3.6/site-packages/_pytest/python.py", line 182 in pytest_pyfunc_call
  File "/home/circleci/.local/lib/python3.6/site-packages/pluggy/callers.py", line 187 in _multicall
  File "/home/circleci/.local/lib/python3.6/site-packages/pluggy/manager.py", line 87 in <lambda>
```

chhzh123, 2020-05-22:
Same error experienced.

Hecmay, 2020-05-22:
The problematic test function is pinpointed here: https://github.com/cornell-zhang/heterocl/blob/master/hlib/tests/test_keras.py#L320

Not fully sure what's going wrong here. I will comment out this function and fix it latter.

PR #220 (2020-05-23): [Test] Fix the 3rd party package version mismatch issue
Should fix #216.
closed 2020-05-23 by Hecmay