cornell-zhang/heterocl #222: Null pointer access causes SegFault when using `reuse_at`

chhzh123, 2020-05-27:
I use the test case in [`test_schedule_memory.py`](https://github.com/cornell-zhang/heterocl/blob/611b2744df6b83a9b69f369a78bb9f993564a971/tests/test_schedule_memory.py), and it fails when executing `hcl.build`.

```python
def test_reuse_blur_x():
    hcl.init()
    A = hcl.placeholder((10, 10))
    B = hcl.compute((10, 8), lambda y, x: A[y, x] + A[y, x+1] + A[y, x+2])
    s = hcl.create_schedule([A, B])
    RB = s.reuse_at(A, s[B], B.axis[1])
    f = hcl.build(s)
```

It generates a segmentation fault without other output messages.

After carefully tracing, I find this bug happens in [L305](https://github.com/cornell-zhang/heterocl/blob/611b2744df6b83a9b69f369a78bb9f993564a971/tvm/src/pass/generate_reuse_buffer.cc#L305) of `generate_reuse_buffer.cc`, where `alloc` is trying to access its `buffer_var` member. However, `alloc` is a null pointer, causing invalid memory access, i.e. SegFault.

[L300-L305](https://github.com/cornell-zhang/heterocl/blob/611b2744df6b83a9b69f369a78bb9f993564a971/tvm/src/pass/generate_reuse_buffer.cc#L300-L305) is shown below. No conditional check is enforced to the returned `alloc` pointer, making the program crash later.
```cpp
        const AttrStmt* attr_alloc = node->body.as<AttrStmt>();
        const Allocate* alloc = attr_alloc->body.as<Allocate>();
        Array<Expr> normal_shape;
        for (int i = reuse_shape.size()-1; i >= 0; i--)
          normal_shape.push_back(reuse_shape[i]);
        shape_map_[alloc->buffer_var.get()] = normal_shape;
```

I have tested this case on two computers, and both of them failed. The configurations are:
* My laptop: Windows 10 + WSL1.0 (Ubuntu18.04 LTS) + LLVM9.0
* A cloud server: Ubuntu18.04 LTS + LLVM6.0

chhzh123, 2020-05-27:
@Hecmay said the problem results from the [ScheduleOps](https://github.com/cornell-zhang/heterocl/blob/master/python/heterocl/tvm/build_module.py#L342) pass.

The lowered IR after this command is shown below.
```
// attr [extern(placeholder0.reuse, 0x29fb1f0)] realize_scope = ""
realize placeholder0.reuse() {
  // attr [[buffer(placeholder0, 0x2ccaa50), Tensor(shape=[10, 10], op.name=placeholder0)]] buffer_bind_scope = tvm_tuple(0, 10, 0, 10)
  // attr [[buffer(placeholder0.reuse, 0x2b2b5e0), Tensor(shape=[], op.name=placeholder0.reuse)]] buffer_bind_scope = tvm_tuple()
  produce placeholder0.reuse {
    // attr [0] extern_scope = 0
    0
  }
  // attr [extern(compute0, 0x2d0fdb0)] realize_scope = ""
  realize compute0([0, 10], [0, 8]) {
    // attr [[buffer(placeholder0, 0x2ccaa50), Tensor(shape=[10, 10], op.name=placeholder0)]] buffer_bind_scope = tvm_tuple(0, 10, 0, 10)
    // attr [[buffer(compute0, 0x2b2cc50), Tensor(shape=[10, 8], op.name=compute0)]] buffer_bind_scope = tvm_tuple(0, 10, 0, 8)
    produce compute0 {
      // attr [0] extern_scope = 0
      for (y, 0, 10) {
        for (x, 0, 8) {
          reuse placeholder0
          // attr [          for (y, 0, 10) {
            // attr [iter_var(y, Range(min=0, extent=10))] loop_scope = y
            for (x, 0, 8) {
              // attr [iter_var(x, Range(min=0, extent=8))] loop_scope = x
              compute0[(x + (y*8))] = int32((int34((int33(placeholder0[(x + (y*10))]) + int33(placeholder0[((x + 1) + (y*10))]))) + int34(placeholder0[((x + 2) + (y*10))])))
            }
          }
] attach_scope = "compute0"
          compute0[(x + (y*8))] = int32((int34((int33(placeholder0[(x + (y*10))]) + int33(placeholder0[((x + 1) + (y*10))]))) + int34(placeholder0[((x + 2) + (y*10))])))
        }
      }
    }
  }
}
```

seanlatias, 2020-06-01:
Solved by #226.

closed 2020-06-01 by seanlatias