cornell-zhang/heterocl #230: `.reuse_at` fails to work with `.to`

chhzh123, 2020-06-04:
The following program using `reuse_at` and `to` causes SegFault.
```python
def test_reuse_blur_x_with_streaming():
    hcl.init()
    A = hcl.placeholder((10, 10), name="A")
    def kernel(A):
        B = hcl.compute((10, 8), lambda y, x: A[y, x] + A[y, x+1] + A[y, x+2],name="B")
        C = hcl.compute((10, 8), lambda y, x: B[y, x], name="C")
        return C
    s = hcl.create_schedule([A], kernel)
    kernel_B = kernel.B
    RB = s.reuse_at(A, s[kernel_B], kernel_B.axis[1])
    target = hcl.platform.zc706
    target.config(compile="vivado_hls",mode="csim")
    s.to(kernel.B, target.xcel)
    s.to(kernel.C, target.host)
    f = hcl.build(s, target)
```

The traceback of gdb is shown below.
```
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
TVM::ir::ReuseBufferInserter::Mutate_ (this=<optimized out>, op=<optimized out>, s=...) at src/pass/generate_reuse_buffer.cc:301
301             const Allocate* alloc = attr_alloc->body.as<Allocate>();
(gdb) backtrace
#0  TVM::ir::ReuseBufferInserter::Mutate_ (this=<optimized out>, op=<optimized out>, s=...) at src/pass/generate_reuse_buffer.cc:301
#1  0x00007fff781fff84 in TVM::ir::<lambda(const Halide::Internal::For*, const Halide::Internal::Stmt&, TVM::ir::IRMutator*)>::operator() (__closure=<optimized out>, m=<optimized out>, s=..., op=<optimized out>)
    at src/pass/ir_mutator.cc:428
#2  std::_Function_handler<Halide::Internal::Stmt(const Halide::Internal::For*, const Halide::Internal::Stmt&, TVM::ir::IRMutator*), TVM::ir::<lambda(const Halide::Internal::For*, const Halide::Internal::Stmt&, TVM::ir::IRMutator*)> >::_M_invoke(const std::_Any_data &, const Halide::Internal::For *&&, const Halide::Internal::Stmt &, TVM::ir::IRMutator *&&) (__functor=..., __args#0=<optimized out>, __args#1=..., 
    __args#2=<optimized out>) at /usr/include/c++/7/bits/std_function.h:302
#3  0x00007fff7820c563 in std::function<Halide::Internal::Stmt (Halide::Internal::For const*, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>::operator()(Halide::Internal::For const*, Halide::Internal::Stmt const&, TVM::ir::IRMutator*) const (__args#2=<optimized out>, __args#1=..., __args#0=<optimized out>, this=<optimized out>) at /usr/include/c++/7/bits/std_function.h:706
#4  TVM::IRFunctor<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>& TVM::IRFunctor<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>::set_dispatch<Halide::Internal::For>(std::function<Halide::Internal::Stmt (Halide::Internal::For const*, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>)::{lambda(TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)#1}::operator()(TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*) const (args#1=<optimized out>, args#0=..., n=..., __closure=<optimized out>)
    at HalideIR/src/tvm/ir_functor.h:108
#5  std::_Function_handler<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*), TVM::IRFunctor<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>& TVM::IRFunctor<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>::set_dispatch<Halide::Internal::For>(std::function<Halide::Internal::Stmt (Halide::Internal::For const*, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>)::{lambda(TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)#1}>::_M_invoke(std::_Any_data const&, TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*&&) (__functor=..., __args#0=..., __args#1=..., __args#2=<optimized out>) at /usr/include/c++/7/bits/std_function.h:302
#6  0x00007fff7813c10a in std::function<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>::operator()(TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*) const (__args#2=<optimized out>, __args#1=..., __args#0=..., this=<optimized out>) at /usr/include/c++/7/bits/std_function.h:706
---Type <return> to continue, or q <return> to quit---
#7  TVM::IRFunctor<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>::operator()(TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*) const (
    this=0x7fff78974b50 <TVM::ir::IRMutator::vtable_stmt()::inst>, n=..., args#0=..., args#1=args#1@entry=0x7fffffff9dd0) at HalideIR/src/tvm/ir_functor.h:75
#8  0x00007fff7813c22d in TVM::ir::IRMutator::Mutate (this=this@entry=0x7fffffff9dd0, stmt=...) at include/tvm/ir_mutator.h:44
#9  0x00007fff78207990 in TVM::ir::IRMutator::Mutate_ (this=this@entry=0x7fffffff9dd0, op=op@entry=0x11ae460, s=...) at src/pass/ir_mutator.cc:144
#10 0x00007fff781d9621 in TVM::ir::ReuseBufferInserter::Mutate_ (this=0x7fffffff9dd0, op=0x11ae460, s=...) at src/pass/generate_reuse_buffer.cc:420
```

If I change the target to `None` or "vhls", the program is able to run. Thus, it seems `reuse_at` cannot capture the new buffer in the modified stage, causing the problem.

Hecmay, 2020-06-06:
Not fully sure what is the root cause here. But by checking the Halide IR fed into `ReuseBufferInserter` pass, it is obvious that the IR is not correct. `A.reuse` stage is not inserted into the right place... 

```c++
produce _top {
  // attr [0] extern_scope = 0
  // attr [A.reuse] storage_scope = "global"
  allocate A.reuse[int32 * 1]
  produce A.reuse {
    // attr [0] extern_scope = 0
    0
  }
  // attr [B] storage_scope = "global"
  allocate B[int32 * 10 * 8]
  produce B {
    // attr [0] extern_scope = 0
    for (y, 0, 10) {
      for (x, 0, 8) {
        reuse A
        produce A.reuse {
          // attr [0] extern_scope = 0
          0
        }
        B[(x + (y*8))] = int32((int34((int33(A[(x + (y*10))]) + int33(A[((x + 1) + (y*10))]))) + int34(A[((x + 2) + (y*10))])))
      }
    }
  }
```
\
I checked the op scheduling results, it maintains the correct order. Namely the `A.reuse` stage is put before the B stage, so that flattening pass should be aware that B stage has a child stage attaching to it...  @seanlatias Do you have any idea what's going wrong here?

Hecmay, 2020-06-06:
This should be fixed in #206. Will merge it into the master after #229.

seanlatias, 2020-06-06:
> This should be fixed in #206. Will merge it into the master after #229.

Can you add a test case? One for reuse_at and one for compute_at.

Hecmay, 2020-06-06:
Sure.

chhzh123, 2020-06-13:
#206 only passes the case above, but still causes SegFault for other cases. As an example, I exchange the compute function of `B` and `C` and get the following program.
```python
def test_reuse_compute2():
    hcl.init()
    A = hcl.placeholder((10, 10),name="A")
    def kernel(A):
        B = hcl.compute((10, 10), lambda y, x: A[y, x], "B")
        C = hcl.compute((10, 8), lambda y, x: B[y, x] + B[y, x+1] + B[y, x+2], "C")
        return C
    s = hcl.create_schedule([A], kernel)
    RB = s.reuse_at(kernel.B._op, s[kernel.C], kernel.C.axis[1])
    target = hcl.platform.zc706
    target.config(compile="vivado_hls",mode="csim")
    s.to(kernel.B, target.xcel)
    s.to(kernel.C, target.host)
    print(hcl.lower(s))
```

Error is shown below.
```
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007fff781d1c6b in TVM::ir::ReuseBufferInserter::Mutate_ (this=0x7fffffffa0e0, op=0x100a180, s=...) at src/pass/generate_reuse_buffer.cc:195
195             size_t ndim = expr_list[0].size();
(gdb) backtrace
#0  0x00007fff781d1c6b in TVM::ir::ReuseBufferInserter::Mutate_ (this=0x7fffffffa0e0, op=0x100a180, s=...) at src/pass/generate_reuse_buffer.cc:195
#1  0x00007fff781f8374 in TVM::ir::<lambda(const Halide::Internal::For*, const Halide::Internal::Stmt&, TVM::ir::IRMutator*)>::operator() (__closure=<optimized out>, m=<optimized out>, s=..., op=<optimized out>)
    at src/pass/ir_mutator.cc:428
#2  std::_Function_handler<Halide::Internal::Stmt(const Halide::Internal::For*, const Halide::Internal::Stmt&, TVM::ir::IRMutator*), TVM::ir::<lambda(const Halide::Internal::For*, const Halide::Internal::Stmt&, TVM::ir::IRMutator*)> >::_M_invoke(const std::_Any_data &, const Halide::Internal::For *&&, const Halide::Internal::Stmt &, TVM::ir::IRMutator *&&) (__functor=..., __args#0=<optimized out>, __args#1=..., 
    __args#2=<optimized out>) at /usr/include/c++/7/bits/std_function.h:302
#3  0x00007fff78204953 in std::function<Halide::Internal::Stmt (Halide::Internal::For const*, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>::operator()(Halide::Internal::For const*, Halide::Internal::Stmt const&, TVM::ir::IRMutator*) const (__args#2=<optimized out>, __args#1=..., __args#0=<optimized out>, this=<optimized out>) at /usr/include/c++/7/bits/std_function.h:706
#4  TVM::IRFunctor<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>& TVM::IRFunctor<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>::set_dispatch<Halide::Internal::For>(std::function<Halide::Internal::Stmt (Halide::Internal::For const*, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>)::{lambda(TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)#1}::operator()(TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*) const (args#1=<optimized out>, args#0=..., n=..., __closure=<optimized out>)
    at HalideIR/src/tvm/ir_functor.h:108
```

zhangzhiru, 2020-06-13:
@Hecmay @seanlatias As I suggested before, please also add test cases for each bug under tests/issues/<issue#>.

Hecmay, 2020-06-20:
The test case should be:

```python
def test_reuse_compute2():
    hcl.init()
    A = hcl.placeholder((10, 10),name="A")
    def kernel(A):
        B = hcl.compute((10, 10), lambda y, x: A[y, x], "B")
        C = hcl.compute((10, 8), lambda y, x: B[y, x] + B[y, x+1] + B[y, x+2], "C")
        return C
    s = hcl.create_schedule([A], kernel)
    target = hcl.platform.zc706
    target.config(compile="vivado_hls",mode="csim")
    B_= s.to(kernel.B, target.xcel)
    B_ = s.reuse_at(B_, s[kernel.C], kernel.C.axis[1])
    s.to(kernel.C, target.host)
    print(hcl.lower(s))
```

The IR fed into the ReuseBufferGenerator IR pass seems to be correct, but the pass failed to locate the reuse pattern somehow. @seanlatias

seanlatias, 2020-06-28:
@Hecmay do you think it is possible to use `reuse_at` before `to`? If you want to use the same order as shown here,  I guess the reason why the reuse pattern is not found is that you did not replace `B` with `B_` in the compute body after applying `to`?

seanlatias, 2020-06-28:
Do I need to use a certain branch of yours? I got a segfault with the program you provided.

Hecmay, 2020-06-29:
@seanlatias Sorry, just saw your reply. 

You can use the "fix" branch in my repo. That's a good suggestion. I can mark the stages when applying .to (without replacing the buffers), and then create new buffers after GenerateReuseBuffer IR pass is finished.

Hecmay, 2021-07-21:
Issue fixed. Test cases added here: https://github.com/cornell-zhang/heterocl/blob/heteroflow/tests/issues/test_issue_230.py

PR #206 (2020-06-13): [API][Utils] Streaming Support Enhancement v2
In this PR, we will introduce:

- [x]  **Enhance the subgraph extraction algorithm for op scheduling**: the current implantation works under many assumptions. The algorithm will check the attaching information between child stage and parent stage, and reconstruct the top-level function. This should fix #230 and #219. Test cases available [here](https://github.com/Hecmay/heterocl/blob/stream_to/tests/test_schedule_stream.py#L529).
- [x] Enhance support for [streaming channel arrays](https://github.com/Hecmay/heterocl/blob/stream_to/samples/systolic_ [... cut]
PR #316 (2020-12-12): [API] Data placement and streaming API rewrite
- [x] Fix the graph hierarchy issue (by reconstructing a call graph from the flattened graph)
- [x] Moved the code transformation logic to the end of the lowering process. Should fix #160 and fix #262 and fix #207 
- [x] Removed the automatically created on-chip buffer when moving data to/from the device. Users can choose from DMA or Stream mode for host-device communication (DMA is enabled by default) Should fix #235.
- [x] Constraints checking (e.g. one-read-one-write e.t.c)
- [x] Fixed the conflicts with `resue_at` (by adding an IR pass to fix  [... cut]
closed 2021-07-21 by Hecmay