cornell-zhang/heterocl #232: Bits of a tensor cannot be accessed in `hcl.select`

chhzh123, 2020-06-05:
The following program causes a Runtime Error when accessing bit 0 of tensor `A` in `hcl.select`.

```python
def bit_test():
    A = hcl.placeholder((8, 8), "A", dtype=hcl.UInt(2))
    B = hcl.placeholder((8, 8), "B", dtype=hcl.UInt(2))
    def kernel(A, B):
        return hcl.compute((8, 8), lambda y, x: 
            hcl.select(x < 4, A[y, x][0], 0), "C")
    s = hcl.create_scheme([A, B], kernel)
    s = hcl.create_schedule_from_scheme(s)
    f = hcl.build(s)
```

Error message is shown below.
```
Traceback (most recent call last):
  File "select_test.py", line 40, in <module>
    bit_test()
  File "select_test.py", line 34, in bit_test
    f = hcl.build(s)
  File "/home/chz/heterocl/python/heterocl/api.py", line 318, in build
    return _build(schedule.sch, new_inputs, target=target, name=name, stmt=stmt)
  File "/home/chz/heterocl/python/heterocl/tvm/build_module.py", line 565, in build
    stmt=stmt)
  File "/home/chz/heterocl/python/heterocl/tvm/build_module.py", line 375, in lower
    stmt = ir_pass.RewriteUnsafeSelect(stmt)
  File "/home/chz/heterocl/python/heterocl/tvm/_ffi/function.py", line 280, in my_api_func
    return flocal(*args)
  File "/home/chz/heterocl/python/heterocl/tvm/_ffi/_ctypes/function.py", line 183, in __call__
    ctypes.byref(ret_val), ctypes.byref(ret_tcode)))
  File "/home/chz/heterocl/python/heterocl/tvm/_ffi/base.py", line 66, in check_call
    raise TVMError(py_str(_LIB.TVMGetLastError()))
heterocl.tvm._ffi.base.TVMError: [19:29:09] include/tvm/ir_functor_ext.h:153: Do not have a default for GetBit
```

However, if I remove `hcl.select`, the program is able to run.

seanlatias, 2020-06-07:
Should be solved by #237.

chhzh123, 2020-06-07:
Yes... This issue is solved, but #193 happens again. After removing the rewrite_unsafe_select pass, the data type casting before some variables is removed, causing synthesis error.

zhangzhiru, 2020-06-07:
We are supposed to add a regression test for each bug fix. Did we do that for #193? Going forward, let's do mention where the new test case is added before closing an issu.e

seanlatias, 2020-06-07:
Yes. We did that for #193. I did modify the test to reflect our latest change. However, it seems that the modification is not always true. Let me do more verification.

seanlatias, 2020-06-07:
I bring back the old test for testing #193 in #238. This should ensure the generated VHLS code after removing the `rewrite_unsafe_select` pass work.

seanlatias, 2020-06-09:
@chhzh123 please see if this is solved. Thanks.

chhzh123, 2020-06-09:
Yes, it's fixed. Thanks!

closed 2020-06-09 by chhzh123