cornell-zhang/heterocl #234: "TypeError: 'Call' object is not subscriptable" when intermediate result is accessed

chhzh123, 2020-06-06:
The following code passing an intermediate result to a function and trying to access its bits causes a Runtime Error.
```python
def popcnt(num):
    out = hcl.scalar(0, "popcnt")
    with hcl.for_(0, 8) as i:
        out.v += num[i]
    return out.v

def test():
    hcl.init()
    A = hcl.placeholder((3,), dtype=hcl.UInt(8), name="A")
    B = hcl.placeholder((3,), dtype=hcl.UInt(8), name="B")
    out = hcl.compute((3,), lambda x: popcnt(A[x] ^ B[x]))
    s = hcl.create_schedule([A, B, out])
    f = hcl.build(s)
```

Traceback is shown below.
```
Traceback (most recent call last):
  File "nested.py", line 42, in <module>
    test()
  File "nested.py", line 33, in test2
    out = hcl.compute((3,), lambda x: popcnt(A[x] ^ B[x]))
  File "/home/chz/heterocl/python/heterocl/compute_api.py", line 295, in compute
    tensor = compute_body(name, lambda_ivs, fcompute, shape, dtype, attrs=attrs)
  File "/home/chz/heterocl/python/heterocl/compute_api.py", line 127, in compute_body
    ret = fcompute(*var_list)
  File "nested.py", line 33, in <lambda>
    out = hcl.compute((3,), lambda x: popcnt(A[x] ^ B[x]))
  File "nested.py", line 26, in popcnt
    out.v += num[i]
TypeError: 'Call' object is not subscriptable
```

This error only occurs when an expression is passed into a function. If an item is passed in, like `popcnt(A[x])`, the program is able to work.

chhzh123, 2020-06-06:
Similarly, this doesn't work.
```python
rb = hcl.reduce_axis(0, 8, name="rb")
out = hcl.compute((3,), lambda x: hcl.sum((A[x] ^ B[x])[rb],axis=rb))
```

seanlatias, 2020-06-07:
Should be solved by #236.

chhzh123, 2020-06-07:
Yes, it solves my problem. Thanks!

closed 2020-06-07 by chhzh123