cornell-zhang/heterocl #248: Problems with running the SODA backend

zhanqiu-hu, 2020-06-18:
There are errors due to SODA while running the file tutorial_09_stencil.py.

zhangzhiru, 2020-06-18:
@Blaok Yuze, can you take a look?

seanlatias, 2020-06-18:
@Blaok, that tutorial is out-dated. Could you update it? Also, can you propagate the errors of SODA to HeteroCL? You can execute the existing tutorial and see that there is indeed an error. However, that error is not reflected in HeteroCL. Only the error message is printed out. We should also terminate the HeteroCL program with the same error message.

Blaok, 2020-06-18:
@zhangzhiru Sure, the problem is caused by the recent changes in the SODA backend. I'll update the tutorial and propagate errors from SODA properly.

@seanlatias Any suggestions on propagating the errors back to HeteroCL? Maybe `exit(EXIT_FAILURE)` if SODA returns non-zero?

PR #249 (2020-06-21): [Backend][SODA] Update tutorial and propagate SODA error
Fix #248.
closed 2020-06-21 by seanlatias