cornell-zhang/heterocl #251: Conda forge package unavailable for OSX

rachitnigam, 2020-06-22:
I tried to install HeteroCL using the instructions here: http://heterocl.csl.cornell.edu/doc/install_from_prebuilt.html#install-with-conda

But ran into this error:
```
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - heterocl

Current channels:

  - https://conda.anaconda.org/cornell-zhang/osx-64
  - https://conda.anaconda.org/cornell-zhang/noarch
  - https://repo.anaconda.com/pkgs/main/osx-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/osx-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

This is presumably because there is no OSX build for the package. Would it be possible to add such a build?

zhangzhiru, 2020-07-05:
This issue seems to have fell through the cracks. 
@Hecmay in our installation guide, can we provide a list of operating systems that we can currently support?

Hecmay, 2020-07-06:
We can only provide the conda package for linux distributions. The building process targeting OSX can easily fail with conda's built-in g++ compiler and packaging tools. 

Users can still build HeteroCL from source on OSX, but packaging it with conda for OSX is hard. I cannot find a way to do that after trying out all possible solutions I found on the Internet.