cornell-zhang/heterocl #263: Installation problem

Leo1726, 2020-07-29:
I followed the instruction from http://heterocl.csl.cornell.edu/doc/install_from_src.html
During the build process, I got the following error:
```C
../heterocl/tvm/dmlc-core/include/dmlc/./parameter.h:997:25: warning: catching polymorphic type ‘const class std::out_of_range’ by value [-Wcatch-value=]
     } catch (const std::out_of_range) {
                         ^~~~~~~~~~~~
../heterocl/tvm/dmlc-core/include/dmlc/./parameter.h: In member function ‘virtual void dmlc::parameter::FieldEntry<double>::Set(void*, const string&) const’:
../heterocl/tvm/dmlc-core/include/dmlc/./parameter.h:1022:25: warning: catching polymorphic type ‘const class std::out_of_range’ by value [-Wcatch-value=]
     } catch (const std::out_of_range) {
                         ^~~~~~~~~~~~
src/codegen/llvm/codegen_llvm.cc: In member function ‘virtual void TVM::codegen::CodeGenLLVM::VisitStmt_(const Halide::Internal::Print*)’:
src/codegen/llvm/codegen_llvm.cc:1436:111: error: request for member ‘getCallee’ in ‘((TVM::codegen::CodeGenLLVM*)this)->TVM::codegen::CodeGenLLVM::module_.std::unique_ptr<llvm::Module>::operator->()->llvm::Module::getOrInsertFunction(llvm::StringRef(((const char*)"printf")), call_ftype)’, which is of pointer type ‘llvm::Constant*’ (maybe you meant to use ‘->’ ?)
   llvm::Function* printf_call = llvm::cast<llvm::Function>(module_->getOrInsertFunction("printf", call_ftype).getCallee());
                                                                                                               ^~~~~~~~~
```
* Environment: Centos7; llvm (svn-trunk@346828)

seanlatias, 2020-07-29:
Which LLVM version is that?

Leo1726, 2020-07-29:
> Which LLVM version is that?

8.0.0svn (a special svn version)

seanlatias, 2020-07-29:
Can you try 6.0?

seanlatias, 2020-07-29:
Or you need to upgrade it to 9.0 I believe.

Leo1726, 2020-07-29:
> Can you try 6.0?
> Or you need to upgrade it to 9.0 I believe.

OK, I would have a try, thanks.

Leo1726, 2020-07-29:
I built llvm 6.0 and set its llvm-config path in HCL's Makefile.config. It works.
I guess llvm8.0svn is not compatible.

chhzh123, 2023-02-26:
As we have released a [new version](https://github.com/cornell-zhang/heterocl/releases/tag/v0.5) of HeteroCL that entirely moves the compilation stack to MLIR, issues related to the original HalideIR flow will be closed. Please feel free to check out the latest HeteroCL and see if it works.

closed 2023-02-26 by chhzh123