cornell-zhang/heterocl #273: Partition pragma on the same array

chhzh123, 2020-09-03:
It's unable to partition the same array along different dimensions.
```python
def conv2():
    A = hcl.placeholder((6, 6), "A")
    F = hcl.placeholder((3, 3), "F")

    def kernel(A, F):
        r = hcl.reduce_axis(0, 3)
        c = hcl.reduce_axis(0, 3)
        return hcl.compute((4, 4),
                lambda y, x: hcl.sum(A[y+r, x+c] * F[r, c], axis=[r, c]), "B")

    s = hcl.create_schedule([A, F], kernel)
    LB = s.reuse_at(A, s[kernel.B], kernel.B.axis[0], "LB")
    WB = s.reuse_at(LB, s[kernel.B], kernel.B.axis[1], "WB")
    s.partition(F, hcl.Partition.Cyclic, factor=3, dim=2)
    s.partition(F, hcl.Partition.Cyclic, factor=3, dim=1) # different dimensions
    s[kernel.B].pipeline(kernel.B.axis[1])
    
    target = hcl.platform.zc706
    target.config(compile="vivado_hls",mode="csyn")
    f = hcl.build(s, target=target)
    hcl_A = hcl.asarray(np.random.randint(0, 10, A.shape))
    hcl_F = hcl.asarray(np.random.randint(0, 10, F.shape))
    hcl_B = hcl.asarray(np.zeros((4, 4)))
    f(hcl_A, hcl_F, hcl_B)
```
Got the following error.
```
heterocl.tvm._ffi.base.TVMError: [13:00:46] src/schedule/schedule_reorder.cc:420: Check failed: !subgraph_op_names.count(name) F.partitioned
```
If I change the partition commands to
```python
F_ = s.partition(F, hcl.Partition.Cyclic, factor=3, dim=2)
s.partition(F_, hcl.Partition.Cyclic, factor=3, dim=1) # different dimensions
```
I obtain a wrong piece of code.
```cpp
void test(bit32 A[6][6], bit32 F[3][3], bit32 B[4][4])
    bit32 F_partitioned;
    #pragma HLS array_partition variable=F_partitioned cyclic dim=1 factor=3
    bit32 F_partitioned_partitioned;
    #pragma HLS array_partition variable=F cyclic dim=2 factor=3
```

Hecmay, 2020-09-03:
The first check means that we do not allow two stages to have the same name. If you can rename the second partition with a unique name, it should pass the first check.

chhzh123, 2020-09-03:
> The first check means that we do not allow two stages to have the same name. If you can rename the second partition with a unique name, it should pass the first check.

I think in this case we only have one stage named `B`? And we do not provide APIs to name the partitions.

seanlatias, 2020-09-03:
We do not have a way two apply partition twice right now.

seanlatias, 2020-09-03:
It should be an easy fix though.

seanlatias, 2020-09-03:
Wait, I think we can partition twice. It's just that the dataflow analysis need to improve.
@Hecmay, you can take a look into this and just ignore the repeated partition stages.

seanlatias, 2020-09-03:
Or maybe I can just name them differently. Either way should work.

seanlatias, 2020-09-03:
@chhzh123 can you try this branch? https://github.com/seanlatias/heterocl/tree/loopcount

I also removed the loopcount=1 logic.

Hecmay, 2020-09-11:
@chhzh123  @seanlatias fixed. I used Hongzheng's code as the recession test case. Please see the `test_issue_273.py`.

Hecmay, 2021-07-21:
Issues fixed already. Test cases added here: https://github.com/cornell-zhang/heterocl/blob/heteroflow/tests/issues/test_issue_273.py

closed 2021-07-21 by Hecmay