cornell-zhang/heterocl #286: Support for hls::stream in VHLS backend

Hecmay, 2020-09-22:
The FIFO is implemented as an annotated multi-dim array in VHLS backend:
```c++
int fifo[100];
#pragma HLS STREAM variable=fifo depth=100
```

However, this makes it hard to declare multiple FIFO channels (which is needed when combining tiling and streaming).
```c++
hls::stream FIFO_arr[O];
for (outer, O)
    for (inner, I)
        FIFO_arr[outer].write(data);
```

zhangzhiru, 2020-09-24:
Why is it difficult to declare multiple FIFO channels using the array-based coding style?

Hecmay, 2020-09-24:
@zhangzhiru I meant currently in heterocl it is hard to do so. It is not a problem in HLS.

zhangzhiru, 2020-09-24:
I see. Can you provide a high-level reason why this is hard in HeteroCL?

Hecmay, 2020-09-24:
@zhangzhiru This is not hard to implement in HeteroCL. This issue is opened just for bookkeeping, and it will be supported in the next release.

zhangzhiru, 2020-10-02:
@chhzh123 you're saying you've already modified the codegen logic to generate hls::stream?

chhzh123, 2020-10-03:
> @chhzh123 you're saying you've already modified the codegen logic to generate hls::stream?

Yes, I use a simple but tricky way to generate the stream reads and writes -- If `_pipe` or `_channel` is in variable names, then the varible is supposed to be a stream buffer. This method is used by Shaojie at the very beginning, where only the VHLS codegen needs to be changed. However, to distinguish a normal buffer and a stream buffer, it would be better to register a new IR node, like `StreamAllocate`. I think @Hecmay has done this and make the stream buffers generation robust in his PR?

Hecmay, 2020-10-03:
Yes. This has been done in #265.

Hecmay, 2021-07-21:
This has already been fixed. Test cases: https://github.com/cornell-zhang/heterocl/blob/heteroflow/tests/test_schedule_stream.py#L340

PR #316 (2020-12-12): [API] Data placement and streaming API rewrite
- [x] Fix the graph hierarchy issue (by reconstructing a call graph from the flattened graph)
- [x] Moved the code transformation logic to the end of the lowering process. Should fix #160 and fix #262 and fix #207 
- [x] Removed the automatically created on-chip buffer when moving data to/from the device. Users can choose from DMA or Stream mode for host-device communication (DMA is enabled by default) Should fix #235.
- [x] Constraints checking (e.g. one-read-one-write e.t.c)
- [x] Fixed the conflicts with `resue_at` (by adding an IR pass to fix  [... cut]
closed 2021-07-21 by Hecmay