cornell-zhang/heterocl #289: Weird segfault when adding two bit slice objects

seanlatias, 2020-09-27:
The following code leads to a segfault

```python
def kernel(A):
  return hcl.compute(A.shape, lambda x: A[x][0:7] + A[x][8:15])
```

The segfault only happens when the bit slices are the exact numbers as shown in the above code. Other numbers don't seem to create a segfault.

seanlatias, 2020-09-27:
We already know that the segfault happens in the `Simplify` pass, but the cause is still unknown.

seanlatias, 2020-09-27:
The data type of `A` also doesn't seem to matter.

zhangzhiru, 2020-09-27:
Just to confirm -- [0:7] means we are extracting the first 8 bits, right? 
Last Wed, I saw Xiangyi/Alga use [0:8] and [8:16]

seanlatias, 2020-09-27:
No, we will be extracting the first 7 bits, which is the same as Python's convention. For Python, the range is upper-bound exclusive.

zhangzhiru, 2020-09-27:
Can you take a look at the bit slicing support in other HW-oriented Python DSLs, like PyMTL and PyRTL? See if they are following this convention.