cornell-zhang/heterocl #292: Insider Backend in HCL

Hecmay, 2020-09-30:
## Programming model in Insider 
Insider's ISC driver and firmware automatically intercepts the input file information when the `vread()` is invoked by the host program. Namely, the input is moved to the storage unit (SU), and users do not need to create explicit write operations in the host program. On the FPGA side, there are three channels for reading large-size input from SU, writing output to host and getting small-size params from host. The FPGA kernel interface is fixed (i.e. we cannot add new channels aside from the existing three). 

## HCL Interface for Insider Backend
To accommodate the insider backend, we need to provide appropriate abstraction for it. I will prefer to use the fluent programming style (i.e. chained .to() APIs) to express the logic. For the host-device interface, we have to pack multiple inputs into a single struct, and unpack it on the FPGA. 

Here is an example of KNN to show the proposed interface. We added a new option (i.e. `vfile=/path/`) to specify the path of the virtual file, and then mention that the input will be consumed by the ISC (i.e. FPGA accelerator). The rest APIs are basically the same as we already have.

```python
def knn(training_data, input_feat):
    # algorithms defined here
    # return predication
    return pred
    
s = hcl.create_schedule()
p = hcl.platform.insider

# 1. Input data is moved to storage unit 
# instead of FPGA, and automatically streamed into FPGA in runtime 
s.to(training_data, p.drive, vfile="/mnt/user/training.dat").to(p.FPGA)

# 2. Input features (small size params) 
# are send from CPU to FPGA through FIFOs
s.to(input_feat, p.FPGA)

# 3. Prediction result streamed back to host
s.to(knn.pred, p.host)
```

zhangzhiru, 2020-10-01:
@Hecmay how do we simulate this code? We won't be able to put data under /mnt/user on the CPU server that runs the HCL simulation. Does Insider have a CPU simulation mode?

Hecmay, 2020-10-01:
Yeah. Insider has CSIM and co-SIM modes. We can provide an interface to invoke Insider's simulation tools. 
```python
p = hcl.platform.insider
p.config(mode="csim|cosim")
```

zhangzhiru, 2020-10-04:
>> s.to(training_data, p.drive, vfile="/mnt/user/training.dat").to(p.FPGA)

We need to think critically about what is the right programming interface here. 
What is the first .to() doing? Isn't the file already stored on disk? 

Also, how do we specify the data type stored in training.dat? I assume we have a fixed-width channel between the drive and the accelerator. In that case, does Insider do the type conversion automatically?

Hecmay, 2020-10-04:
In the HCL user end, the input training data is in-memory (e.g. from numpy array), we need to dump it into the file (stored in the disk). If the data is already in the file, I think we can also define it as
```python
s.to(training_data, p.FPGA, vfile="/mnt/user/training.dat")
```

About the data type, this is the required interface data type in Insider. Basically the data in `training,dat` is packed into 512 bit and passed to FPGA. Users need to do the unpacking manually in the kernel code. 
```c++
struct APP_Data {
  ap_uint<512> data;
  unsigned short len;
  bool eop;
};
```

In the provided example, I did not see any floating or fixed-point data type used in for the interface.

Hecmay, 2020-10-07:
Made some changes to the previous proposal:
1. Use placeholder to represent the file descriptor, so that the `.to` primitive looks more concise 
2. Add an interface to configure the BW and interconnect latency of Insider.

```python
def knn(training_data, input_feat):
    # algorithms defined here
    # return predication
    return pred

# 1. Define placeholders. The in-storage data should be stored in the file before hand
training_data = hcl.placeholder((size,), source="/home/users/input.txt")
params = hcl.placeholder((size, ))

# 2. Configure the simulation parameters 
p = hcl.platform.insider
p.config(bandwidth=12, latency=300)

# 3. Move data. Large data is moved from disk to FPGA
# while smaller params for moved from host to FPGA
s = hcl.create_schedule(kernel, [training_data, params])
s.to(training_data, p.drive).to(p.FPGA)
s.to(params, p.FPGA)
```