cornell-zhang/heterocl #299: Avoiding importing too many dependency libraries when importing hlib

Hecmay, 2020-10-12:
When importing `hlib`, many other packages are imported by default (e.g. TVM, keras), which makes the importing process really slow (around 5s). 

I do not know whether this is related to Debjit's issue. He tried to install TVM separately in the OS, but pip always complains TVM not found. Maybe there is some potential conflicts between TVM and HCL.TVM in pip?

seanlatias, 2020-10-12:
Although the conclusion is not true (tvm conflicts with hcl.tvm), I agree we can reduce the number of dependency libraries needed. If some package is not found, usually that just means the environment issue.

Hecmay, 2020-10-12:
This is kind od an unrelated issue. Debjit said he installed TVM using the `.circleci/install_tvm.sh`, but in the site-packages folder, he can only find an empty `tvm.egg` folder. That's wired. we can check later.

chhzh123, 2023-02-26:
As we have released a [new version](https://github.com/cornell-zhang/heterocl/releases/tag/v0.5) of HeteroCL that entirely moves the compilation stack to MLIR, issues related to the original HalideIR flow will be closed.

closed 2023-02-26 by chhzh123