cornell-zhang/heterocl #302: TVM Argument Binding Failed for 512-bit UInt Datatype

zzzDavid, 2020-10-24:
## Problem Description
FlexCNN uses 512-bit global input output bus but implementing it in HeteroCL causes TVM error in `src/pass/arg_binder.cc`.

`hcl.UInt(512)` seems to generate `uint0` buffer which causes TVM arg binding problem.

**Error Message**:
```bash
heterocl.tvm._ffi.base.TVMError: [16:41:10] src/pass/arg_binder.cc:84: Check failed: arg->dtype == value->dtype (uint0 vs. uint32) Argument global_cin Buffer bind data type mismatch

Stack trace returned 10 entries:
[bt] (0) 0   libhcl.dylib                        0x00000001136facae dmlc::StackTrace() + 254
[bt] (1) 1   libhcl.dylib                        0x00000001136faa5f dmlc::LogMessageFatal::~LogMessageFatal() + 47
[bt] (2) 2   libhcl.dylib                        0x00000001138860af TVM::ir::ArgBinder::BindBuffer(TVM::Buffer const&, TVM::Buffer const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) + 575
[bt] (3) 3   libhcl.dylib                        0x00000001139aa6e7 TVM::ir::StorageFlattener::HandleBufferBindScope(Halide::Internal::AttrStmt const*) + 4423
[bt] (4) 4   libhcl.dylib                        0x000000011399dad5 TVM::ir::StorageFlattener::Mutate_(Halide::Internal::AttrStmt const*, Halide::Internal::Stmt const&) + 1941
[bt] (5) 5   libhcl.dylib                        0x00000001138f8e05 std::__1::__function::__func<TVM::ir::$_1, std::__1::allocator<TVM::ir::$_1>, Halide::Internal::Stmt (Halide::Internal::AttrStmt const*, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>::operator()(Halide::Internal::AttrStmt const*&&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*&&) + 21
[bt] (6) 6   libhcl.dylib                        0x00000001138f83d1 std::__1::__function::__func<TVM::IRFunctor<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>& TVM::IRFunctor<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>::set_dispatch<Halide::Internal::AttrStmt>(std::__1::function<Halide::Internal::Stmt (Halide::Internal::AttrStmt const*, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>)::'lambda'(TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*), std::__1::allocator<TVM::IRFunctor<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>& TVM::IRFunctor<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>::set_dispatch<Halide::Internal::AttrStmt>(std::__1::function<Halide::Internal::Stmt (Halide::Internal::AttrStmt const*, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>)::'lambda'(TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>, Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>::operator()(TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*&&) + 49
[bt] (7) 7   libhcl.dylib                        0x000000011374daec TVM::IRFunctor<Halide::Internal::Stmt (TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*)>::operator()(TVM::NodeRef const&, Halide::Internal::Stmt const&, TVM::ir::IRMutator*) const + 348
[bt] (8) 8   libhcl.dylib                        0x00000001138675fb TVM::ir::IRMutator::Mutate(Halide::Internal::Stmt) + 59
[bt] (9) 9   libhcl.dylib                        0x00000001139aa812 TVM::ir::StorageFlattener::HandleBufferBindScope(Halide::Internal::AttrStmt const*) + 4722
``` 

## Repeat the Error

HeteroCL version: [Hecmay/heterocl:fix](https://github.com/Hecmay/heterocl/tree/b1faa504f645c9edcc4bba4db1805c53115aac5a)

Code: [zzzDavid:heterocl/samples/flexcnn/flexcnn.py](https://github.com/zzzDavid/heterocl/blob/867812ccc0787b890bad2cfad8b22bb863298a2b/samples/flexcnn/flexcnn.py)
(Needs samples/flexcnn/kernel/*.cpp)

```bash
$ python samples/flexcnn/flexcnn.py
```

## Possible Cause

Argument binding supports only up to 128-bit

seanlatias, 2020-10-24:
We only support up to 255 bits in HeteroCL right now.

zhangzhiru, 2020-10-24:
@seanlatias  what prevents us from allowing wider integer? At the very least, we need to prompt error message instead of letting the tool crash.

seanlatias, 2020-10-24:
I think I mentioned this before, we use 8-bit to store the total bitwidth and thus we can only support up to 255 bits.

seanlatias, 2020-10-24:
If you think this one has a higher priority then I'll fix this first.

zhangzhiru, 2020-10-24:
Yes, we should fix this issue since we already have a relatively simple solution in mind.
I thought this one is different since Niansong is talking about the global input/output.

seanlatias, 2020-10-24:
No, according to his code, he is just generating the code without running the cpu simulation. In this case, we are not limited to numpy and thus we should be able to generate code for larger bitwidth for the global input/output.

zhangzhiru, 2020-10-24:
Let's be sure to prompt error message. We also need to create test case to check the message.

seanlatias, 2020-10-28:
@zzzDavid should be fixed by #303. Let me know otherwise.