cornell-zhang/heterocl #304: CodeGen crashed when print zero-sized buffer allocation

Hecmay, 2020-10-29:
After merging with new PR (#303), I tried to run the flexcnn.py. The lowering worked fine, but the code generator complains that some allocate IR node has zero size. Here is the error msg:

```shell
heterocl.tvm._ffi.base.TVMError: [22:04:59] src/codegen/opencl/codegen_xocl_host.cc:182: Check failed: constant_size > 0 (0 vs. 0) Can only handle constant size stack allocation for now. Buffer compute5 has 0 stack size.

Stack trace returned 10 entries:
[bt] (0) /work/shared/users/phd/sx233/heterocl/tvm/lib/libhcl.so(dmlc::StackTrace()+0x40) [0x7fec31e426c0]
[bt] (1) /work/shared/users/phd/sx233/heterocl/tvm/lib/libhcl.so(dmlc::LogMessageFatal::~LogMessageFatal()+0x2b) [0x7fec31e42e4b]
[bt] (2) /work/shared/users/phd/sx233/heterocl/tvm/lib/libhcl.so(TVM::codegen::CodeGenXOCLHost::VisitStmt_(Halide::Internal::Allocate const*)+0x7eb) [0x7fec32148b4b]
[bt] (3) /work/shared/users/phd/sx233/heterocl/tvm/lib/libhcl.so(TVM::IRFunctor<void (TVM::NodeRef const&, TVM::ir::StmtFunctor<void (Halide::Internal::Stmt const&)>*)>::operator()(TVM::NodeRef const&, TVM::ir::StmtFunctor<void (Halide::Internal::Stmt const&)>*) const+0x10e) [0x7fec3206d93e]
[bt] (4) /work/shared/users/phd/sx233/heterocl/tvm/lib/libhcl.so(TVM::codegen::CodeGenC::VisitStmt_(Halide::Internal::AttrStmt const*)+0xb0) [0x7fec320c1e70]
[bt] (5) /work/shared/users/phd/sx233/heterocl/tvm/lib/libhcl.so(TVM::IRFunctor<void (TVM::NodeRef const&, TVM::ir::StmtFunctor<void (Halide::Internal::Stmt const&)>*)>::operator()(TVM::NodeRef const&, TVM::ir::StmtFunctor<void (Halide::Internal::Stmt const&)>*) const+0x10e) [0x7fec3206d93e]
[bt] (6) /work/shared/users/phd/sx233/heterocl/tvm/lib/libhcl.so(TVM::IRFunctor<void (TVM::NodeRef const&, TVM::ir::StmtFunctor<void (Halide::Internal::Stmt const&)>*)>::operator()(TVM::NodeRef const&, TVM::ir::StmtFunctor<void (Halide::Internal::Stmt const&)>*) const+0x10e) [0x7fec3206d93e]
[bt] (7) /work/shared/users/phd/sx233/heterocl/tvm/lib/libhcl.so(TVM::IRFunctor<void (TVM::NodeRef const&, TVM::ir::StmtFunctor<void (Halide::Internal::Stmt const&)>*)>::operator()(TVM::NodeRef const&, TVM::ir::StmtFunctor<void (Halide::Internal::Stmt const&)>*) const+0x10e) [0x7fec3206d93e]
[bt] (8) /work/shared/users/phd/sx233/heterocl/tvm/lib/libhcl.so(TVM::codegen::CodeGenXOCLHost::VisitStmt_(Halide::Internal::Allocate const*)+0x565) [0x7fec321488c5]
[bt] (9) /work/shared/users/phd/sx233/heterocl/tvm/lib/libhcl.so(TVM::IRFunctor<void (TVM::NodeRef const&, TVM::ir::StmtFunctor<void (Halide::Internal::Stmt const&)>*)>::operator()(TVM::NodeRef const&, TVM::ir::StmtFunctor<void (Halide::Internal::Stmt const&)>*) const+0x10e) [0x7fec3206d93e]
```

The test case is available here: https://github.com/Hecmay/heterocl/blob/fix/samples/flexcnn/flexcnn.py

Hecmay, 2020-10-29:
@seanlatias after I comment out the check `CHECK_GT(const_size > 0)`, the `ExtractIndices()` function will throw out another error.

seanlatias, 2020-10-29:
Why do we have zero-sized allocate node in the first place? This looks more like a bug in flexcnn to me.

zhangzhiru, 2020-10-29:
Regardless, another pass to prompt error messages. Supposedly, the compiler should never crash.

Hecmay, 2020-10-29:
Actually I checked all the `hcl.compute`, and they all look good to me. Not sure. We can take a closer look later

Hecmay, 2021-09-23:
This should have been fixed in https://github.com/cornell-zhang/heterocl/pull/418. 
Test case added in tests/issues/test_issue_304.py. Please reopen the issue if the error still exists.

closed 2021-09-23 by Hecmay