cornell-zhang/heterocl #310: Bitcast Operation Support

zzzDavid, 2020-11-09:
This issue is to document bitcast operation which is not directly exposed to user.

## Description
FlexCNN uses 512-bit bus to transport packed 32-bit raw bits, which is casted to 32-bit floating point data upon computation.  In this case, we need support for bitcast operation. 

## Note
`hcl.cast()` is cast by value, meaning uint `0010` would be converted to `2.0f`. Bitcast operation is available in HeteroCL but not directly exposed to user: check [this example](https://github.com/cornell-zhang/heterocl/blob/bdb87b01cbdf613fe746d25dd949e18cd4942ecf/python/heterocl/tensor.py#L166).

seanlatias, 2021-01-25:
@zzzDavid I remember you implemented something? You can contribute if you'd like.

zzzDavid, 2021-01-26:
Yes, I implemented a `bitcast()` API in `python/heterocl/compute_api.py` and corresponding tests. 

It's in the heteroflow branch now:

https://github.com/zzzDavid/heterocl/blob/heteroflow/python/heterocl/compute_api.py#L971

seanlatias, 2021-01-26:
Could you extract that part of the code and create a separate PR? Thanks.

zzzDavid, 2021-01-27:
Sure, the new PR is: https://github.com/cornell-zhang/heterocl/pull/330

closed 2021-07-09 by seanlatias