cornell-zhang/heterocl #311: Tensor index casting with imperative for loops can lead to LLVM errors

seanlatias, 2020-11-14:
With the existing imperative for loops, we will transform the loop variable to make it always start from zero. For instance

```python
with hcl.for_(0, 10) as i:
  with hcl.for_(i, 10) as j:
    A[i][j] = ...
```

This becomes

```
for i in (0, 10):
  for j in (0, (10-i)):
   A[i][j+i] = ...
```

Since we automatically perform some casting to keep the accuracy, sometimes the tensor index becomes
```
A[i][(int33)j + (int33)i]
```

With the current LLVM backend, the index can only be of int32. Thus, the above code can produce LLVM errors.

zhangzhiru, 2020-11-14:
Is there a description of the rules we use to do the casting? 
Why do we have to add these implicit casts on the intermediate variables? HLS tool will do it automatically for us. I thought we only need to handle the ones that we have to declare.

seanlatias, 2020-11-14:
Because we want the CPU simulation results to be the same as HLS's results.

seanlatias, 2020-11-14:
Later on, when we extend our tool to allow users to define their own data types, they can define the casting behavior whatever they want. For now, the casting behavior is set to be the same as Xilinx HLS's behavior.

seanlatias, 2020-11-14:
Also, LLVM is very picky in that you cannot add two variables with different data types. Thus, we still need to do some sort of casting.

zhangzhiru, 2020-11-14:
Again, Is there a description of the rules we use to do the casting?
What's the solution to fix this bug?

seanlatias, 2020-11-14:
The rule is exactly the same as in the Xilinx user manual for all operations (e.g., add/sub/mul). We don't handle div. The solution is to remove the casting to make LLVM happy.

seanlatias, 2020-11-14:
In general, we do not expect users to use ap data types for indices. If that's the case, we should return a warning.

zhangzhiru, 2020-11-14:
I did not know that the Xilinx user manual describes how the HLS tool casts the intermediate variables? Can you send me the link?
It's not clear to me what we have to cast i and j in the first place? Both of them are declared using the same type.

zhangzhiru, 2020-11-14:
> The solution is to remove the casting to make LLVM happy.

Under what circumstances do we not cast? Only for array indices? We have to clearly describe our solution. Otherwise, we may open another can of worms.

seanlatias, 2020-11-14:
https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug902-vivado-high-level-synthesis.pdf

For instance, Page 527 defines the width of the output after addition.

seanlatias, 2020-11-14:
In our case, since both i and j are int32, the output width would be int33 according to the manual.

zhangzhiru, 2020-11-14:
> In our case, since both i and j are int32, the output width would be int33 according to the manual.

I understood this part well. The question is when do we add the explicit type cast in LLVM and the HLS code. I don't think it's described anywhere. Do we add it for all intermediate variables? When do we drop it, etc.?

In this example, would it help if we add another cast (int64) for the array indices? The HLS tool will optimize it away.

seanlatias, 2020-11-14:
So here is an example of the usual case, say we have o = i + j, all of them are int32, then this is what we generate

```
o = (int32) (int33(i) + int33(j))
```

As you can see, we will finally cast back to LHS at the end. Since our goal is to produce the same result as HLS, we need to explicitly cast the intermediate results. For now, the generated code also contains the casting, which is proved to work fine with so many examples we have right now.

For array indices, we can definitely do that as you proposed.

seanlatias, 2020-11-14:
Otherwise, how should I compute this `int32 = int3 + int5`? Do we cast `int3` to `int5` or cast both `int3` and `int5` to `int6`? According to the manual, the latter should be the solution. The reason we need to cast is LLVM does not allow two operands with different data types.

zhangzhiru, 2020-11-14:
> int32 o = (int32) (int33(i) + int33(j))
The cast to int33 seems totally unnecessary to me. HLS tool does this automatically. 

>> int32 = int3 + int5
This is also fine for HLS without any explicit casts.

For the LLVM code gen, when we have z = x OP y. We cast x and y to the max_width_of(x, y), then cast it back to the width of z. For array indices, the width is of z is 32b or 64b, depending on the target CPU architecture.

seanlatias, 2020-11-14:
> For the LLVM code gen, when we have z = x OP y. We cast x and y to the max_width_of(x, y), then cast it back to the width of z. For array indices, the width is of z is 32b or 64b, depending on the target CPU architecture.

But would this kind of casting results in different results as HLS? Because in the manual the return bitwidth should be max(x, y) + 1.

zhangzhiru, 2020-11-14:
Right, we should do max_width+1 for the intermediate result. 
But again, no need to generate cast in the HLS code.

seanlatias, 2020-11-14:
Or do you think it's fine if we do not produce the same results as HLS?

seanlatias, 2020-11-14:
ok, I'll open another issue for that. We can move the casting to LLVM level. But I'll still create a quick fix for Debjit to continue on developing the benchmarks.

zhangzhiru, 2020-11-14:
Sounds good. It's fine that we generate redundant HLS code for now. 
But we should clean it up. In addition, we don't need to generate warning if ap_int is used for array indexing.