cornell-zhang/heterocl #328: [Refactor] Reference argument usage in C++

seanlatias, 2021-01-26:
We should enforce ourselves to do the following (according to Google C++ style):

```c++
void (const A& a); //good
void (A* a); //good
void (A& a); //bad
```

This is also a common coding style (at least TVM is also requiring this). We should create a PR purely for cleaning up this usage. Tagging the current contributors.

@Hecmay @paldebjit @zzzDavid @Sibylau

seanlatias, 2021-01-26:
This applies not only to function arguments but also class constructors.

chhzh123, 2023-02-26:
As we have released a [new version](https://github.com/cornell-zhang/heterocl/releases/tag/v0.5) of HeteroCL that entirely moves the compilation stack to MLIR, issues related to the original HalideIR flow will be closed.

closed 2023-02-26 by chhzh123