cornell-zhang/heterocl #335: An extendable and elegant device target interface embracing both FPGA and ASIC

Sibylau, 2021-02-05:
For FPGA, there's a host and a device code. The programmers can write HeteroFlow code in the following style:
```python
target = hcl.platform.aws_f1
s = hcl.create_schedule([in_list], kernel)
s.to(input, target.xcel, mode)
s.to(output, target.host, mode)
```
or if the target is a custom one, the coding style is:
```python
config = {
  "host": hcl.dev.host("vendor"),
  "xcel": hcl.dev.xcel("vendor")}
target = hcl.platform.custom(config)
s = hcl.create_schedule([in_list], kernel)
s.to(input, target.xcel, mode)
s.to(output, target.host, mode)
```
But in ASIC design flow, the terminology is quite different. We write DUT code and a testbench to test the functionality of DUT. And usually, in HLS code targeting ASIC designs, there're at least one header files gluing them. I wonder what will be an elegant and extendable way for HeteroCL to support both FPGA and ASIC. My current thoughts about the coding style for ASIC design in mind is shown below. Any discussion is welcomed.
```python
target = hcl.platform.asic("vendor")
s = hcl.create_schedule([in_list], kernel)
s.to(input, target.dut, mode)
s.to(output, target.testbench, mode)
```

Hecmay, 2021-02-06:
Thanks, @Sibylau. 

Here is a follow-up about the user interface for programmers to define their own custom platforms. This is one of the possible alternatives we can use, instead of having the dictionary input. 

 
```python
target = hcl.platform.custom
target.add(hcl.dev.fpga.xilinx("vu9p"))
target.add(hcl.dev.cpu.intel("v5"))

```

In the case of ASIC devices, we may do
```python
target = hcl.platform.custom
target.add(hcl.dev.asic("vendor"))
```

Sibylau, 2021-02-16:
Thank you @Hecmay . Your proposed interface seems also good to me. Tag @zhangzhiru for additional comments

Hecmay, 2021-02-25:
A few more thoughts about the `Platform` and `Device` class in HCL. 

- A platform in HCL is a collection of synthesis tools and devices. The devices in a platform are connected in a pre-defined way. 
- A platform provides information to HCL runtime to decide which code generator and synthesis tool to use. To make it more useful, we can also use it for accelerator deployment. 

E.g., if we want to compile and run the bitstream on AWS. The `compile()` and `run()` functions can be modified by programmers. This would make the runtime more extendable and flexible. 

```python
p = hcl.Platform.AWS_F1(image_id="xxxxx")
f = hcl.build(s, target=p)

f.compile(instance="t2")
f.run(args, instance="f1")
```