cornell-zhang/heterocl #339: Update Conda Installation Instructions

sudo-michael, 2021-02-21:
After creating a conda environment and installing heterocl in the following manner

`conda install -c cornell-zhang heterocl -c conda-forge`

there appears to be a missing module for future
```
(hcl-test) mike@mai:~$ python3
Python 3.8.8 | packaged by conda-forge | (default, Feb 20 2021, 16:22:27) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import heterocl
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mike/.local/lib/python3.8/site-packages/heterocl-0.3-py3.8.egg/heterocl/__init__.py", line 2, in <module>
    from .api import *
  File "/home/mike/.local/lib/python3.8/site-packages/heterocl-0.3-py3.8.egg/heterocl/api.py", line 5, in <module>
    from .tvm.build_module import build as _build, lower as _lower
  File "/home/mike/.local/lib/python3.8/site-packages/heterocl-0.3-py3.8.egg/heterocl/tvm/__init__.py", line 13, in <module>
    from . import schedule
  File "/home/mike/.local/lib/python3.8/site-packages/heterocl-0.3-py3.8.egg/heterocl/tvm/schedule.py", line 6, in <module>
    from ..devices import Device, DevMediaPair
  File "/home/mike/.local/lib/python3.8/site-packages/heterocl-0.3-py3.8.egg/heterocl/devices.py", line 5, in <module>
    from future.utils import with_metaclass
ModuleNotFoundError: No module named 'future'
>>> 
```

This can be fixed by adding the following to the instructions
`conda install -c anaconda future`

seanlatias, 2021-02-21:
@sudo-michael thanks for the suggestions! @Hecmay, maybe we should just include `future` into `setup.py`?

Hecmay, 2021-02-21:
Sure. I can update the conda package to include the package.

Hecmay, 2021-03-10:
Fixed in #332. We do not depend on future pkg anymore.

seanlatias, 2021-03-23:
@sudo-michael please give it a try and let us know if it's still not working. Thanks.

sudo-michael, 2021-03-23:
It's working now. Thanks!

PR #357 (2021-04-14): [Testing] Add test cases for issues related to data movement 
## Issues on deprecated features 
- [x] Removed metaclass in devices.py. Should fix #339 
- [x] HCL runtime switched from shared memory to JSON. Should fix #241 and fix #290

## Issues already fixed in #332 
- [x] Duplicate data movement commands will be ignored automatically. Should fix #270 

## Issues/requests for new features 
- [x] Added config attribute in tool class. Users can parameterize EDA tool options. Should fix #326 
- [x] Added tests for dataflow() primitive. Should fix #317 
- [x] Added tests for burst memory loading [... cut]
closed 2021-04-01 by sudo-michael