cornell-zhang/heterocl #351: Use MXNET==1.6.0 in CI testing

Hecmay, 2021-03-22:
Our CI setup scripts automatically install latest MXNET 1.8.0 (released yesterday), which will cause some errors in the testing:

```shell
circleci@0d8b008dea90:~/project/samples/lenet$ python lenet_main.py 
Using TensorFlow backend.
Traceback (most recent call last):

  File "lenet_main.py", line 54, in <module>
    import mxnet as mx

  File "/home/circleci/.local/lib/python3.7/site-packages/mxnet/__init__.py", line 23, in <module>
    from .context import Context, current_context, cpu, gpu, cpu_pinned

  File "/home/circleci/.local/lib/python3.7/site-packages/mxnet/context.py", line 23, in <module>
    from .base import classproperty, with_metaclass, _MXClassPropertyMetaClass

  File "/home/circleci/.local/lib/python3.7/site-packages/mxnet/base.py", line 351, in <module>
    _LIB = _load_lib()

  File "/home/circleci/.local/lib/python3.7/site-packages/mxnet/base.py", line 342, in _load_lib
    lib = ctypes.CDLL(lib_path[0], ctypes.RTLD_LOCAL)

  File "/usr/local/lib/python3.7/ctypes/__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)

OSError: libopenblas.so.0: cannot open shared object file: No such file or directory
```

@seanlatias can you please specify an older version? I tried in the interactive shell. MXNET 1.6.0 should work fine.

seanlatias, 2021-03-22:
Sure. Go ahead and use that version. Please create a separate PR for that.

seanlatias, 2021-03-23:
Close with #352.

closed 2021-03-23 by seanlatias