cornell-zhang/heterocl #353: Modulo operation fails on operands greater than 128-bit in LLVM backend

Sibylau, 2021-03-23:
When using the LLVM backend, modulo operation `%` fails with integers longer than 128 bit.
The test case is linked here: 
```python
import heterocl as hcl
import numpy as np

N_HASH_FUNC = 5
HASH_RANGE = 10

def func_wrapper(dtype):
    hcl.init(dtype)
    a=hcl.placeholder((N_HASH_FUNC,), dtype=dtype)
    b=hcl.placeholder((N_HASH_FUNC,), dtype=dtype)
    sketch = hcl.placeholder((HASH_RANGE,), dtype=dtype)
    key=hcl.placeholder((1,),dtype=dtype)
    def hash_index(sketch, a, b, key):
        index = hcl.compute(a.shape, lambda i: (a[i]*key[0] + b[i])%HASH_RANGE)
        with hcl.for_(0, N_HASH_FUNC) as i:
            sketch[index[i]] += 1
    
    s = hcl.create_schedule([sketch, a, b, key], hash_index)
    f = hcl.build(s, target="llvm")

    hcl_sketch = hcl.asarray(np.zeros(N_HASH_FUNC*HASH_RANGE),dtype=dtype)
    hcl_a = hcl.asarray(np.random.randint(10, size=(N_HASH_FUNC,)), dtype=dtype)
    hcl_b = hcl.asarray(np.random.randint(10, size=(N_HASH_FUNC,)), dtype=dtype)
    hcl_key = hcl.asarray(np.random.randint(10, size=(1,)), dtype=dtype)

    f(hcl_sketch, hcl_a, hcl_b, hcl_key)

if __name__ == "__main__":
    func_wrapper(hcl.Int(64))
```

compiles with error:
`LLVM ERROR: Unsupported library call operation!`

seanlatias, 2021-05-20:
It seems LLVM doesn't support long bit-width division. I'm trying to see if there's any solution. In the worst case, we might need to implement the division as a series of substractions.

seanlatias, 2021-05-20:
I also found the same error in another public repo: https://github.com/ziglang/zig/issues/1534.
@zhangzhiru any suggestion? @zhenkun can you help me tag Jeremy?

zhenkun, 2021-05-21:
> I also found the same error in another public repo: [ziglang/zig#1534](https://github.com/ziglang/zig/issues/1534).
> @zhangzhiru any suggestion? @zhenkun can you help me tag Jeremy?

@jcasas00