cornell-zhang/heterocl #366: Installation issues

willsharpless, 2021-04-30:
Hi, I've been having installation issues all day, but can't figure out why. I thought it might be my anaconda installation which was a little funky so I uninstalled it and textbook installed the manager but still run into this same issue. I'm using zsh on Mac OS Catalina 10.15.7

Not sure why it can't see heterocl as I can search on anaconda.org and find it.

```
(hcl-env) willsharpless@MacBook-Pro-2 ~ % conda install -c cornell-zhang heterocl -c conda-forge
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - heterocl

Current channels:

  - https://conda.anaconda.org/cornell-zhang/osx-64
  - https://conda.anaconda.org/cornell-zhang/noarch
  - https://conda.anaconda.org/conda-forge/osx-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/osx-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/osx-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

seanlatias, 2021-04-30:
Can you try installing from scratch instead of using `conda install`? We haven't tested on macOS before.

@Hecmay is it because we don't have conda support for macOS?

willsharpless, 2021-04-30:
Yeah,  so after cloning with
`git clone https://github.com/cornell-zhang/heterocl.git`
I then cd to heterocl and try to make but run into the following error (last lines)
`(hcl-env) willsharpless@MacBook-Pro-2 ~ % cd heterocl
(hcl-env) willsharpless@MacBook-Pro-2 heterocl % ls
CPPLINT.cfg	Makefile.config	docker		pkgs		tests
LICENSE		README.md	docs		python		tutorials
Makefile	conda		hlib		samples		tvm
(hcl-env) willsharpless@MacBook-Pro-2 heterocl % make -j8
/Library/Developer/CommandLineTools/usr/bin/make -C pkgs
/bin/sh: wget: command not found
tar: Error opening archive: Failed to open 'cmake-3.10.2-Linux-x86_64.tar.gz'
rm: cmake-3.10.2-Linux-x86_64.tar.gz: No such file or directory
mv: rename cmake-3.10.2-Linux-x86_64 to cmake: No such file or directory
make[2]: *** [/Users/willsharpless/heterocl/build/pkgs/cmake/build/.built-cmake] Error 1
make[1]: *** [all] Error 1
make: *** [build-pkgs] Error 2`

I just followed the from scratch installation instructions to a t, looks like they were written for a linux os. Note I do have xcode on here though some cli functions are available to me.

seanlatias, 2021-04-30:
I think the error comes from this message `wget: command not found`. You need to install `wget` first.

willsharpless, 2021-04-30:
More progress after installing wget

```
(base) willsharpless@MacBook-Pro-2 heterocl % make -j8
/Library/Developer/CommandLineTools/usr/bin/make -C pkgs
--2021-04-30 16:50:57--  https://cmake.org/files/v3.10/cmake-3.10.2-Linux-x86_64.tar.gz
Resolving cmake.org... 66.194.253.19
Connecting to cmake.org|66.194.253.19|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 34221307 (33M) [application/x-gzip]
Saving to: 'cmake-3.10.2-Linux-x86_64.tar.gz'

cmake-3.10.2-Linux- 100%[===================>]  32.64M  1.43MB/s    in 24s     

2021-04-30 16:51:22 (1.34 MB/s) - 'cmake-3.10.2-Linux-x86_64.tar.gz' saved [34221307/34221307]

Downloading LLVM ...
--2021-04-30 16:51:24--  http://releases.llvm.org/6.0.0/llvm-6.0.0.src.tar.xz
Resolving releases.llvm.org... 2a04:4e42:45::561, 199.232.34.49
Connecting to releases.llvm.org|2a04:4e42:45::561|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://releases.llvm.org/6.0.0/llvm-6.0.0.src.tar.xz [following]
--2021-04-30 16:51:24--  https://releases.llvm.org/6.0.0/llvm-6.0.0.src.tar.xz
Connecting to releases.llvm.org|2a04:4e42:45::561|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 25296048 (24M) [application/octet-stream]
Saving to: 'llvm-6.0.0.src.tar.xz'

llvm-6.0.0.src.tar. 100%[===================>]  24.12M  22.3MB/s    in 1.1s    

2021-04-30 16:51:26 (22.3 MB/s) - 'llvm-6.0.0.src.tar.xz' saved [25296048/25296048]

cd /Users/willsharpless/heterocl/build/pkgs/llvm/build; \
	/Users/willsharpless/heterocl/build/pkgs/cmake/build/cmake/bin/cmake -G "Unix Makefiles" -DCMAKE_INSTALL_PREFIX=/Users/willsharpless/heterocl/build/pkgs/install -DCMAKE_BUILD_TYPE=Release /Users/willsharpless/heterocl/build/pkgs/llvm/src
/bin/sh: /Users/willsharpless/heterocl/build/pkgs/cmake/build/cmake/bin/cmake: cannot execute binary file
make[2]: *** [/Users/willsharpless/heterocl/build/pkgs/llvm/build/config.status] Error 126
make[1]: *** [all] Error 1
make: *** [build-pkgs] Error 2
```

seanlatias, 2021-04-30:
I guess in this case you'll need to install your `cmake`. The cmake we install for you is for Linux and not OSX compatible.

Hecmay, 2021-04-30:
I had some difficulties creating conda package for HCL on MacOS due to some dependency conflicts. But I can give a second try later.

willsharpless, 2021-05-01:
I got it to work! For future macOS users,

- conda install wget and cmake

then running make -j8 in the heteroCL dir works. Note, it took a suprisingly long time (still only 30 min or so)  compared to the linux installation.

Python script using heterocl is running properly.

closed 2021-05-01 by willsharpless

melodylylin, 2021-07-21:
Hi, I'm also on Mac, and I've tried install from source, however, I get this error,

/Library/Developer/CommandLineTools/usr/bin/make -C tvm
cd HalideIR; make lib/libHalideIR.a DMLC_CORE_PATH=../dmlc-core; cd /Users/LiYuLin/heterocl/tvm
make[2]: `lib/libHalideIR.a' is up to date.
cd python; \
	python setup.py install --user; \
	cd ../hlib/python; \
	python setup.py install --user;
Traceback (most recent call last):
  File "setup.py", line 1, in <module>
    from setuptools import setup, find_packages
ImportError: No module named setuptools
Traceback (most recent call last):
  File "setup.py", line 1, in <module>
    from setuptools import setup, find_packages
ImportError: No module named setuptools
make: *** [build-hcl] Error 1

And I've tried pip install setuptools, pip3 install setuptools, and conda install setuptools, but still have this error.

seanlatias, 2021-07-21:
You need to make sure Python can find the path where you install the packages. If you are using conda, make sure you are in the right environment. Not sure if this [article](https://realpython.com/lessons/why-cant-python-find-my-modules/) is helpful.

melodylylin, 2021-07-22:
> You need to make sure Python can find the path where you install the packages. If you are using conda, make sure you are in the right environment. Not sure if this [article](https://realpython.com/lessons/why-cant-python-find-my-modules/) is helpful.

I got it works, thanks for replying! Somehow my setuptools is in python3 not python, so I changed the commands in makefile.