cornell-zhang/heterocl #368: TensorSlice doesn't have `shape` property

zhenkun, 2021-05-13:
`A = hcl.compute((2,10), lambda i,j: 0)`
`A[1].shape`

will result in this:
_Error in sys.excepthook:
Traceback (most recent call last):
  File ".../heterocl-0.1-py3.7.egg/heterocl/debug.py", line 70, in hcl_excepthook
    print("\33[1;34m[HeteroCL Error]\33[0m" + value.message)
AttributeError: 'TensorError' object has no attribute 'message'

Original exception was:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File ".../heterocl-0.1-py3.7.egg/heterocl/tensor.py", line 164, in __getattr__
    "Cannot access attribute if type is not struct")
heterocl.debug.TensorError: [Tensor] Cannot access attribute if type is not struct_

seanlatias, 2021-05-13:
I will add this property but note that a slice of tensor cannot be used alone. In other words, you cannot write something like `A[0]+A[1]`.

seanlatias, 2021-05-13:
@zhenkun please check #369 and see if that fixes your issue. Thanks!

zhenkun, 2021-05-13:
@seanlatias , thanks for the quick update! I will try it, and let you know.

zhenkun, 2021-05-14:
works now! Thanks.

closed 2021-05-14 by zhenkun