cornell-zhang/heterocl #370: `domain` of type `hcl.Int(64)` in `hcl.mutate` will trigger LLVM assertion (core dump)

zhenkun, 2021-05-13:
`import numpy as np
import heterocl as hcl
def comp(A):
    d = hcl.cast(hcl.Int(64), 5)    # imagine this value was computed and of type Int64
    #B = hcl.compute((d,), lambda i: A[i] + 5, "B") # this is okay
    def doit(i):
        A[i] += 5
    hcl.mutate((d,), lambda i: doit(i))             # this crashes

A = hcl.placeholder((10,), "A")
A_ = hcl.asarray(np.zeros(A.shape), dtype=hcl.Int(16))

s = hcl.create_schedule([A], comp)
m = hcl.build (s)   # fault`

will result in:
`python: /usr/lib/llvm-9/include/llvm/IR/Instructions.h:1151: void llvm::ICmpInst::AssertOK(): Assertion `getOperand(0)->getType() == getOperand(1)->getType() && "Both operands to ICmp instruction are not of the same type!"' failed.
Aborted (core dumped)`

seanlatias, 2021-05-14:
Weird. Somehow I cannot reproduce this issue. Maybe that's because of the LLVM version. Let me try out a newer LLVM version. I'm using LLVM 6.0 right now.

zhenkun, 2021-05-14:
I'll pull the latest see if the problem goes away.

zhenkun, 2021-05-14:
@seanlatias @Hecmay I tried the latest. the problem went away! Sorry about that.

closed 2021-05-14 by zhenkun