cornell-zhang/heterocl #377: runtime assertion support

zhenkun, 2021-06-03:
similar to `hcl.print`, it would be nice to have a `hcl.assert` that will check the condition at runtime.
Right now python `assert` is only valid at elaboration time.

seanlatias, 2021-06-04:
Great idea. What kind of behavior do you expect? I would expect the HeteroCL program to stop but not the entire Python program. With this setting, users can run 10 HCL programs sequentially even if one of them fails the assertion.

zhenkun, 2021-06-04:
I agree with you. Just stopping the HeteroCL program makes more sense. If it is possible, it would also be nice to support string formatting. For example, `hcl.assert(a>b, "a and b are different: a=%d, b=%d\n", a, b)`, or something similar. But if it's too hard, we can always call `hcl.print` explicitly.

seanlatias, 2021-07-09:
Solved in #385

seanlatias, 2021-07-09:
@zhenkun please give it a try. The only difference with your example is we need to use `hcl.assert_` with an underscore.

closed 2021-07-09 by seanlatias